realsenseai/realsense-ros #1399: CMake error - Could not find package configuration file for "librealsense2"

Sammysedo, 2020-09-17:
Hello,

I have ROS2 Foxy built from source on Windows 10 and I'm having trouble building the ros2_intel_realsense package. I built librealsense2 from source (since I previously had compatibility issues with the binary version on Python 3.8), using instructions from [here](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md) and [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python). The python libraries are now working:
```
Python 3.8.3 (tags/v3.8.3:6f8c832, May 13 2020, 22:37:02) [MSC v.1924 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyrealsense2 as rs
>>>                           
```

However, I am not able to build the ros2 package, instead I am getting a CMake error:
```
C:\sample_ws>colcon build --merge-install
Starting >>> cv_bridge
Starting >>> message_filters
Starting >>> camera_calibration_parsers
Starting >>> image_geometry
Finished <<< image_geometry [7.77s]
Starting >>> realsense_camera_msgs
Finished <<< camera_calibration_parsers [9.42s]
Starting >>> camera_info_manager
Finished <<< cv_bridge [15.2s]
Starting >>> opencv_tests
Finished <<< camera_info_manager [5.88s]
Starting >>> vision_opencv
Finished <<< opencv_tests [3.95s]
Finished <<< vision_opencv [4.49s]
Finished <<< message_filters [20.2s]
Starting >>> image_transport
Finished <<< realsense_camera_msgs [13.9s]
Finished <<< image_transport [7.33s]
Starting >>> image_common
Starting >>> realsense_ros2_camera
Finished <<< image_common [1.91s]
--- stderr: realsense_ros2_camera
CMake Error at CMakeLists.txt:65 (find_package):
  By not providing "Findlibrealsense2.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "librealsense2", but CMake did not find one.

  Could not find a package configuration file provided by "librealsense2"
  with any of the following names:

    librealsense2Config.cmake
    librealsense2-config.cmake

  Add the installation prefix of "librealsense2" to CMAKE_PREFIX_PATH or set
  "librealsense2_DIR" to a directory containing one of the above files.  If
  "librealsense2" provides a separate development package or SDK, be sure it
  has been installed.

---
Failed   <<< realsense_ros2_camera [2.17s, exited with code 1]

Summary: 10 packages finished [30.2s]
  1 package failed: realsense_ros2_camera
  1 package had stderr output: realsense_ros2_camera
```

It seems that CMake is unable to locate my build of librealsense2. I am unable to locate `librealsense2Config.cmake` or `librealsense2-config.cmake` anywhere in my build directory. I have rebuilt several times with ALL_BUILD and it's always the same error. I have tried setting `librealsense2_DIR` to the root directory but it seems to fail. Any ideas?

- Python 3.8
- CMake 3.18.2
- ROS2 Foxy
- Windows 10

doronhi, 2020-09-18:
The version for ros eloquent available through this repository builds the package "realsense2_camera", not "realsense_ros2_camera" which is taken from here: https://github.com/intel/ros2_intel_realsense
Very confusing, I know, and will be settled soon.
On this repository we don't have a version for foxy yet.

Sammysedo, 2020-09-18:
@doronhi I realized that fairly soon after I posted this issue and tried to build the eloquent version from this repository. I am, however, facing a similar issue:
```
C:\sample_ws>colcon build --merge-install
Starting >>> cv_bridge
Starting >>> image_transport
...
...
Starting >>> realsense2_camera
Starting >>> image_common
Finished <<< image_common [7.12s]
--- stderr: realsense2_camera
CMake Warning at CMakeLists.txt:103 (find_package):
  By not providing "Findrealsense2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "realsense2", but CMake did not find one.

  Could not find a package configuration file provided by "realsense2"
  (requested version 2.38.1) with any of the following names:

    realsense2Config.cmake
    realsense2-config.cmake

  Add the installation prefix of "realsense2" to CMAKE_PREFIX_PATH or set
  "realsense2_DIR" to a directory containing one of the above files.  If
  "realsense2" provides a separate development package or SDK, be sure it has
  been installed.

CMake Error at CMakeLists.txt:105 (message):

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

---
Failed   <<< realsense2_camera [8.76s, exited with code 1]

Summary: 9 packages finished [2min 5s]
  1 package failed: realsense2_camera
  1 package had stderr output: realsense2_camera
  1 package not processed
```

After setting `realsense2_DIR` to my librealsense2 build directory (as suggested) I am getting the following error:

```
--- stderr: realsense2_camera
CMake Error at C:/librealsense/build/CMakeFiles/Export/lib/cmake/realsense2/realsense2Targets.cmake:95 (message):
  The imported target "realsense2::realsense-file" references the file

     "C:/librealsense/build/CMakeFiles/Export/lib/realsense-filed.lib"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "C:/librealsense/build/CMakeFiles/Export/lib/cmake/realsense2/realsense2Targets.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  C:/librealsense/build/realsense2Config.cmake:35 (include)
  CMakeLists.txt:103 (find_package)

---
Failed   <<< realsense2_camera [4.34s, exited with code 1]
Aborted  <<< image_common [1.70s]

Summary: 8 packages finished [17.9s]
  1 package failed: realsense2_camera
  1 package aborted: image_common
  1 package had stderr output: realsense2_camera
  1 package not processed
```

So far I have tried master and development branches of `librealsense` & eloquent and development branches of this repository with no avail.

doronhi, 2020-09-20:
I never tried it with windows but usually you shouldn't set realsense2_DIR or anything else. 
Both the master and development branches of librealsense2 should be fine but it should be easier if you just installed librealsense2 the way it is described in the [librealsense2 repository](https://github.com/IntelRealSense/librealsense)

If you use eloquent there is no point is using any other branch of this repository.

RealSenseSupport, 2020-09-27:
@Sammysedo Any other questions with doronhi's reply? Looking forward to your reply. Thanks!

Sammysedo, 2020-09-28:
@RealSenseSupport Hi and thank you for your comment. Actually I do have additional questions:

I've managed to (sort of) tackle most of the issues I've stumbled upon so far. Now, I'm getting errors for unresolved externals:

```
  Checking Build System
  Building Custom Rule C:/sample_ws/src/realsense-ros/realsense2_node/CMakeLists.txt
     Creating library C:/sample_ws/build/realsense2_node/Release/realsense2_node.lib and object C:/sample_ws/build/realsense2_node/Release/realsense2_node.exp
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_get_librealsense_exception_type referenced in function "public: __cdecl rs2::error::error(struct rs2_error *)" (??0error@rs2@@QEAA@PEAUrs2_error@@@Z) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_get_failed_function referenced in function "public: __cdecl rs2::error::error(struct rs2_error *)" (??0error@rs2@@QEAA@PEAUrs2_error@@@Z) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_get_failed_args referenced in function "public: __cdecl rs2::error::error(struct rs2_error *)" (??0error@rs2@@QEAA@PEAUrs2_error@@@Z) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_get_error_message referenced in function "public: __cdecl rs2::error::error(struct rs2_error *)" (??0error@rs2@@QEAA@PEAUrs2_error@@@Z) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_free_error referenced in function "public: __cdecl rs2::error::error(struct rs2_error *)" (??0error@rs2@@QEAA@PEAUrs2_error@@@Z) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_release_frame referenced in function "public: __cdecl rs2::frame::~frame(void)" (??1frame@rs2@@QEAA@XZ) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_create_sync_processing_block referenced in function "private: class std::shared_ptr<struct rs2_processing_block> __cdecl rs2::asynchronous_syncer::init(void)" (?init@asynchronous_syncer@rs2@@AEAA?AV?$shared_ptr@Urs2_processing_block@@@std@@XZ) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_process_frame referenced in function "public: void __cdecl realsense2_camera::PipelineSyncer::operator()(class rs2::frame)const " (??RPipelineSyncer@realsense2_camera@@QEBAXVframe@rs2@@@Z) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
realsense2_node.obj : error LNK2019: unresolved external symbol rs2_delete_processing_block referenced in function "private: class std::shared_ptr<struct rs2_processing_block> __cdecl rs2::asynchronous_syncer::init(void)" (?init@asynchronous_syncer@rs2@@AEAA?AV?$shared_ptr@Urs2_processing_block@@@std@@XZ) [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
C:\sample_ws\build\realsense2_node\Release\realsense2_node.exe : fatal error LNK1120: 9 unresolved externals [C:\sample_ws\build\realsense2_node\realsense2_node.vcxproj]
Failed   <<< realsense2_node [5.17s, exited with code 1]

Summary: 10 packages finished [25.2s]
  1 package failed: realsense2_node
```

It seems that some of the libraries are not properly linked?

RealSenseSupport, 2020-09-29:
@Sammysedo Noticed that you're using ROS2 foxy which we don't support yet. Is it possible for you to try eloquent which we can support and released recently? https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.1

doronhi, 2020-09-29:
What version of librealsense2 did you install and with what method (From source / debian)?

RealSenseSupport, 2020-10-10:
@Sammysedo Could you please provide feedback to doronhi's previous questions? Thanks!

RealSenseSupport, 2020-11-05:
@Sammysedo Could you please update? Thanks!

RealSenseSupport, 2020-11-11:
@Sammysedo Sorry that we didn't get update from you for weeks. Will close this issue now. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-11-11 by RealSenseSupport