realsenseai/realsense-ros #1405: L515 Gap Between Depth and Color Images

jinbae9875, 2020-09-18:
Hi, below is an experiment I did with L515 and an egg. The edge tracing of the egg is taken from the rgb image corresponding to each of the colorized rgb-aligned depth images.
[image]

0
[image]
1
[image]
2
[image]
3
[image]
4
[image]
5
[image]
6
[image]
7
[image]
8
[image]

As you can see, the shots taken at an angle from the camera (the ones not at the center) have a significant lateral gap between the depth image and its equivalent rgb image. I've tried the same experiment using D435 and Kinect but they don't seem to have the same issue. What do you think is causing this issue? Is anyone else experiencing the same problem?

RealSenseSupport, 2020-10-02:
Hi @jinbae9875 

Is there anyway you can provide the recordings/bag file using the RealSense Viewer of your experiment?  Also provide all streams - IR, Depth, RGB and Confidence in your recording?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-10-21 by RealSenseSupport

zhongxiangzju, 2020-10-30:
Hi, I encountered the same issue which is that the depth image is not aligned well with color image after set "align_depth"               
 to "true" in rs_camera.launch. Is it a bug?