realsenseai/realsense-ros #1407: Realsense d435i crashing occasionally

ivelso, 2020-09-20:
Realsense d435i
Turtlebot3 with intel nuc with ubuntu 16.04
ROS kinetic 1.12.14

The camera can run for a while and then suddenly crash, with the need of disconnecting the camera in order to make it work again.   
The camera frequency is reduced to 15fps in order to receive the images on a remote computer as it was not received any images when using 30fps. Still, a lot of the images are lost but this might be because of transmission between to computers. 
Is it anything to do in order to make the camera more reliable and make it not crash? 

It is also saying  Hardware Notification:USB SCP overflow. The usb should be a usb 3 as it is colored blue but it seems as it is functioning as usb2. The camera is connected using the original cable. 

This is a log of the output from the realsense node: 
 
turtlebot07@turtlebot07:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch 
... logging to /home/turtlebot07/.ros/log/9eda4a1a-fb1c-11ea-94f8-0800274146ea/roslaunch-turtlebot07-15401.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.17:33611/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 30
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: True
 * /camera/realsense2_camera/enable_fisheye2: True
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
[... 193 lines cut ...]
[ INFO] [1600594629.406700055]: Expected frequency for depth = 15.00000
[ INFO] [1600594629.739670855]: Expected frequency for color = 15.00000
[ INFO] [1600594630.101916013]: Expected frequency for aligned_depth_to_color = 15.00000
[ WARN] [1600594630.379274185]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
[ INFO] [1600594630.713101510]: setupStreams...
[ INFO] [1600594630.714938417]: insert Depth to Stereo Module
[ INFO] [1600594630.715058161]: insert Color to RGB Camera
 20/09 19:37:10,749 ERROR [139927832405760] (types.h:307) xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
[ERROR] [1600594630.750359663]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
 20/09 19:37:10,750 ERROR [139927832405760] (context.cpp:427) Exception thrown from user callback handler
[ WARN] [1600594632.381117113]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594634.382881076]: Hardware Notification:USB CAM overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594635.384166479]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
 20/09 19:37:15,739 WARNING [139927644702464] (backend-v4l2.cpp:988) Frames didn't arrived within 5 seconds
[ WARN] [1600594636.385420623]: Hardware Notification:USB CAM overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594637.386635939]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594639.388340851]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
 20/09 19:37:20,744 WARNING [139927644702464] (backend-v4l2.cpp:988) Frames didn't arrived within 5 seconds
[ WARN] [1600594641.390324915]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594643.392083043]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594645.393755879]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
 20/09 19:37:25,746 WARNING [139927644702464] (backend-v4l2.cpp:988) Frames didn't arrived within 5 seconds
[ WARN] [1600594647.395432051]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594649.397314628]: Hardware Notification:USB CAM overflow,1.60059e+12,Error,Hardware Error
[ WARN] [1600594650.398528848]: Hardware Notification:USB SCP overflow,1.60059e+12,Error,Hardware Error

MartyG-RealSense, 2020-09-20:
Hi @ivelso  My first point of advice would be to not use librealsense SDK version 2.34.0 as it had a problem with continuously generating timing errors.   If you wish to keep your currently installed ROS wrapper version 2.2.13 then stepping back one librealsense SDK version earlier to **2.33.1** may be an option, as that SDK can also be matched with ROS wrapper 2.2.13. 

I also note in your log that **align_depth** is set to True.  If you do not need toalign depth to color in your ROs project then setting that option to False may free up some processing.

ivelso, 2020-09-20:
Thank you @MartyG-RealSense    I will try to update the library and see if that helps as I am getting heaps of the error about the time. Will i need to change my launch file for the turtlebot if I update to the newest version of librealsense SDK?

MartyG-RealSense, 2020-09-20:
I would think that if the ROS components of your installation stay the same but the librealsense version is changed then the turtlebot's ROS settings would stay the same.  I have not personally worked with turtlebots myself though so that is an educated guess.  

Updating librealsense versions involves matching that SDK to the ROS wrapper version that is most closely matched to that SDK version.  This information is on the ROS wrapper **Releases** page for each wrapper version.  For example, it is recommended that librealsense SDK **2.38.1** is matched with ROS wrapper version **2.2.17** if using ROS Kinetic or Melodic,

https://github.com/IntelRealSense/realsense-ros/releases

In regard to launch file settings, they remain consistent over the long term whatever the ROS wrapper version is.  Bear in mind though that from wrapper version **2.2.15** onward, the infra, IMU and fisheye are set to False by default, so you should enable these by setting them to True if you need them (either in the roslaunch instruction of within the launch file).

ivelso, 2020-09-24:
I updated the library and the wrapper. getting error that "running with a different librealsense version than the one the wrapper was compiled with! " but it works. 
The camera is still loosing images even when the image rate is 15fps. 
Do you have any other suggestions to improve it? 
I am currently using the aligned_depth_to_color 

turtlebot07@turtlebot07:~$ roslaunch turtlebot3_bringup turtlebot3_robot.launch 
... logging to /home/turtlebot07/.ros/log/ac08a30e-fe04-11ea-854c-0800274146ea/roslaunch-turtlebot07-2532.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.0.17:43533/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 30
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 30
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
[... 99 lines cut ...]
[ INFO] [1600910675.038180776]: Enable PointCloud: Off
[ INFO] [1600910675.038207744]: Align Depth: On
[ INFO] [1600910675.038233246]: Sync Mode: On
[ INFO] [1600910675.038294720]: Device Sensors: 
[ INFO] [1600910675.042071360]: Stereo Module was found.
[ INFO] [1600910675.051565501]: RGB Camera was found.
[ INFO] [1600910675.051908066]: Motion Module was found.
[ INFO] [1600910675.051977544]: (Fisheye, 0) sensor isn't supported by current device! -- Skipping...
[ INFO] [1600910675.052034460]: num_filters: 0
[ INFO] [1600910675.052059468]: Setting Dynamic reconfig parameters.
[INFO] [1600910675.190641]: Calibration End
[ WARN] [1600910675.479903582]: This firmware(v2.0.1) isn't compatible with this software (v1.2.3)
[ WARN] [1600910675.479982705]: You can find how to update its in `FAQ` section(turtlebot3.robotis.com)
[ INFO] [1600910677.644762443]: Done Setting Dynamic reconfig parameters.
[ INFO] [1600910677.645546644]: depth stream is enabled - width: 640, height: 480, fps: 15, Format: Z16
[ INFO] [1600910677.646320623]: color stream is enabled - width: 640, height: 480, fps: 15, Format: RGB8
[ INFO] [1600910677.647974054]: setupPublishers...
[ INFO] [1600910677.733595907]: Expected frequency for depth = 15.00000
[ INFO] [1600910679.518583725]: Expected frequency for color = 15.00000
[ INFO] [1600910681.507173938]: Expected frequency for aligned_depth_to_color = 15.00000
[ INFO] [1600910683.196819570]: setupStreams...
[ INFO] [1600910683.197338642]: insert Depth to Stereo Module
[ INFO] [1600910683.197412905]: insert Color to RGB Camera
[ INFO] [1600910683.205097863]: SELECTED BASE:Depth, 0
[ INFO] [1600910683.974963251]: RealSense Node Is Up!

MartyG-RealSense, 2020-09-24:
I believe that the version error message would be corrected by rebuilding the ROS wrapper **2.2.17** with the SDK **2.38.1** that you already have installed (i.e don't rebuild librealsense, just the RealSense ROS wrapper).  I would guess that the error was caused by building the 2.2.17 wrapper with SDK 2.33.1 installed and then upgrading the SDK version to 2.38.1 afterwards?

Simply building the wrapper again with 2.38.1 installed may make a difference.  I would also suggest adding the command below to your roslaunch instruction to see whether it helps:  It will perform a reset of the camera.

**initial_reset:=true**

MartyG-RealSense, 2020-10-02:
Hi @ivelso   Do you still require assistance with this case, please?  Thanks!

ivelso, 2020-10-03:
Hi @MartyG-RealSense,  I am not able to remove the error of the camera that it is compiled with a different wrapper. I have tried  
$ cd ~/catkin_ws
$ catkin_make
and 
$ catkin clean

but the error is not removed. 

Also the camera performance is not improved. The camera does not crash, but I still do not get images every second. It can go a few seconds between each image containing rgb image and depth image with the same timestamp. The setting for sending images with the same timestamp is on.  

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 30
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 15
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 15
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: True
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 15
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 30
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 30 lines cut ...]
    realsense2_camera_manager (nodelet/nodelet)
  /
    turtlebot3_core (rosserial_python/serial_node.py)
    turtlebot3_diagnostics (turtlebot3_bringup/turtlebot3_diagnostics)
    turtlebot3_lds (hls_lfcd_lds_driver/hlds_laser_publisher)

ROS_MASTER_URI=http://192.168.0.112:11311

process[turtlebot3_core-1]: started with pid [8525]
process[turtlebot3_lds-2]: started with pid [8526]
process[camera/realsense2_camera_manager-3]: started with pid [8527]
process[camera/realsense2_camera-4]: started with pid [8528]
process[turtlebot3_diagnostics-5]: started with pid [8532]
[ INFO] [1601625484.398046345]: Initializing nodelet with 4 worker threads.
[INFO] [1601625484.608400]: ROS Serial Python Node
[ INFO] [1601625484.617051926]: RealSense ROS v2.2.17
[ INFO] [1601625484.617153138]: Built with LibRealSense v2.33.1
[ INFO] [1601625484.617232188]: Running with LibRealSense v2.38.1
[ WARN] [1601625484.617325731]: ***************************************************
[ WARN] [1601625484.617415957]: ** running with a different librealsense version **
[ WARN] [1601625484.617491747]: ** than the one the wrapper was compiled with!   **
[ WARN] [1601625484.617720917]: ***************************************************
[INFO] [1601625484.723156]: Connecting to /dev/ttyACM0 at 115200 baud
[ INFO] [1601625484.922183282]:  
[ INFO] [1601625484.932162498]: Device with serial number 841612071764 was found.

MartyG-RealSense, 2020-10-04:
The next action that I would recommend trying is to set **enable_sync** to False.  This option is not related to depth to color alignment, but instead it "gathers closest frames of different sensors, infrared, color and depth, to be sent with the same timetag".  In most cases that I see, this option is listed as False in the ROS launch log.

ivelso, 2020-10-13:
Hi Marty.  Removing the enable_sync will not send the images with same time stamp making my control not thrust them as I want to ensure that they match in order to find the object and get the distance to the object. If this is not the same it is a chance that the distance is incorrect when the robot is turning. The stream is relatively stable and works good enough for my application now. Thank you for the help.

closed 2020-10-13 by ivelso