realsenseai/realsense-ros #1408: Cannot run

hamsterasu, 2020-09-20:
Hi,
I followed the instructions to install debian package of the ROS wrapper using `sudo apt-get install ros-melodic-realsense2-camera`. From what is said on the README, it says that by installing the debian package,

> This will install both realsense2_camera and its dependents, including librealsense2 library.

So from my understanding of this, I do not have to install anything else, ie. librealsense2.

However, when I run `roslaunch realsense2 rs_camera.launch` while D435i is connected, i get this error:

```
 21/09 00:29:45,089 ERROR [140173394786048] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 21/09 00:29:45,110 WARNING [140173428356864] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1600619385.110816284]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1600619385.110863734]: The requested device with  is NOT found. Will Try again.
[ INFO] [1600619391.122466749]:  
 21/09 00:29:51,127 ERROR [140173403178752] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 21/09 00:29:51,149 WARNING [140173428356864] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1600619391.149189828]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1600619391.149264654]: The requested device with  is NOT found. Will Try again.
```

When I run dmesg, i see this:
```
[ 8160.342817] input: Intel(R) RealSense(TM) Depth Ca as /devices/pci0000:00/0000:00:14.0/usb1/1-12/1-12:1.0/input/input30
[ 8160.344196] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[ 8160.346105] uvcvideo 1-12:1.3: Entity type for entity Processing 7 was not initialized!
[ 8160.346109] uvcvideo 1-12:1.3: Entity type for entity Extension 8 was not initialized!
[ 8160.346112] uvcvideo 1-12:1.3: Entity type for entity Camera 6 was not initialized!
[ 8160.426289] hid-sensor-hub 0003:8086:0B3A.0008: No report with id 0xffffffff found
[ 8160.528972] audit: type=1107 audit(1600619701.599:132): pid=864 uid=103 auid=4294967295 ses=4294967295 msg='apparmor="DENIED" operation="dbus_signal"  bus="system" path="/org/freedesktop/NetworkManager" interface="org.freedesktop.NetworkManager" member="CheckPermissions" name=":1.15" mask="receive" pid=8840 label="snap.spotify.spotify" peer_pid=874 peer_label="unconfined"
                exe="/usr/bin/dbus-daemon" sauid=103 hostname=? addr=? terminal=?'
[ 8165.584965] hid-sensor-hub 0003:8086:0B3A.0008: No report with id 0xffffffff found
[ 8169.170915] hid-sensor-hub 0003:8086:0B3A.0008: No report with id 0xffffffff found
```

When I run `roslaunch realsense2_camera rs_t265.launch` for t265 camera, i also get an error:
```
 21/09 00:59:07,679 ERROR [139917231736576] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 21/09 00:59:07,679 ERROR [139917231736576] (tm-boot.h:42) Failed to open T265 zero interface
[ WARN] [1600621149.688932423]: No RealSense devices were found!
 21/09 00:59:15,700 ERROR [139917231736576] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 21/09 00:59:15,700 ERROR [139917231736576] (tm-boot.h:42) Failed to open T265 zero interface
```

I have attached the full console messages for your viewing.

Full console messages:
[d435i_dmesg.txt](https://github.com/IntelRealSense/realsense-ros/files/5252129/d435i_dmesg.txt)
[d435i_error.txt](https://github.com/IntelRealSense/realsense-ros/files/5252130/d435i_error.txt)
[t265_dmesg.txt](https://github.com/IntelRealSense/realsense-ros/files/5252131/t265_dmesg.txt)
[t265_roslaunch.txt](https://github.com/IntelRealSense/realsense-ros/files/5252132/t265_roslaunch.txt)

RealSenseSupport, 2020-09-23:
@hamsterasu Could you please try if you can run realsenseviewer and start streaming with D435i and T265 successfully? Also what's your platform? And kernel version? Thanks!

hamsterasu, 2020-09-23:
realsense-viewer is not installed. Do I have to? 
My kernel version is 5.4.0-47-generic. Platform is Intel® Core™ i7-10875H CPU

RealSenseSupport, 2020-09-24:
@hamsterasu I reproduced the issue and found one solution. Could you please check if you have 99-realsense-libusb.rules under /etc/udev/rules.d/ ? If not, please copy https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules to /etc/udev/rules.d/  before connect the cameras. After that you can connect the camera and execute the ros launch.

hongtaowu67, 2020-09-26:
It works in my case after putting including the 99-realsense-libusb.rules.

RealSenseSupport, 2020-09-27:
@hongtaowu67  Thanks for your feedback! Glad to hear this works for you.

hamsterasu, 2020-09-27:
@RealSenseSupport Your solution works for me too! Thanks! I am closing this issue

closed 2020-09-27 by hamsterasu

Shunichi09, 2020-10-01:
I got an same problem when I was using docker.
I also tried to solve it by your suggestion (i.e. including the 99-realsense-libusb.rules in /etc/udev/rules.d/.) but it did not work for me...
Are there any solutions ??

I checked realsense-viewer did work and got RGB images.

running command

```
$ roslaunch realsense2_camera rs_camera.launch
```

error message

```
 01/10 16:08:00,157 ERROR [140156020442880] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 01/10 16:08:00,178 WARNING [140156054013696] (rs.cpp:304) null pointer passed for argument "device"
[ WARN] [1601568480.178749109]: Device 1/1 failed with exception: failed to set power state
[ERROR] [1601568480.178776222]: The requested device with  is NOT found. Will Try again.
```

Iko-Git, 2020-10-23:
> @hamsterasu I reproduced the issue and found one solution. Could you please check if you have 99-realsense-libusb.rules under /etc/udev/rules.d/ ? If not, please copy [...]

solved my issue too thanks, but I wonder shouldn't such a file be default installed with realsense package

BellosC, 2020-10-27:
Hello, i am a beginner, and i tried to follow the instructions above but i got this error:
My commands are these:
 konstantinos@Bellos:~$ cd /etc/udev/rules.d/
 konstantinos@Bellos:/etc/udev/rules.d$ git clone https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules
**_and i get this:_**
**fatal: could not create work tree dir '99-realsense-libusb.rules': Permission denied**

camilorequena, 2020-11-02:
> solved my issue too thanks, but I wonder shouldn't such a file be default installed with realsense package

This solved my issue too. Seems to be some kind of bug.

HWeigo, 2020-12-05:
> @hamsterasu I reproduced the issue and found one solution. Could you please check if you have 99-realsense-libusb.rules under /etc/udev/rules.d/ ? If not, please copy [...]

Thanks! Mine is solved too

EricWiener, 2021-01-26:
> @RealSenseSupport I got the same problem when I was using docker.
> I also tried to solve it by your suggestion (i.e. including the 99-realsense-libusb.rules in /etc/udev/rules.d/.) but it did not work for me...
> Are there any solutions ??
> 
> I checked realsense-viewer did work and got RGB-D images, although there was error message.
> Finally, I updated the firmware of realsense to 5.12, I got same error.
> 
> ```
>  01/10 17:41:38,038 ERROR [139670256056064] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
>  01/10 17:41:38,039 ERROR [139670256056064] (fw-update-device.cpp:128) access failed for 8086:adb uid: 2-7-10
>  01/10 17:41:38,039 WARNING [139670256056064] (rs.cpp:304) null pointer passed for argument "device"
> ```
> 
> running command
> 
> ```
> $ roslaunch realsense2_camera rs_camera.launch
> ```
> 
> error message
> 
> ```
> [ INFO] [1601571623.360312063]: RealSense ROS v2.2.16
> [ INFO] [1601571623.360333411]: Built with LibRealSense v2.37.0
> [ INFO] [1601571623.360341901]: Running with LibRealSense v2.37.0
> 01/10 16:08:00,157 ERROR [140156020442880] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
>  01/10 16:08:00,178 WARNING [140156054013696] (rs.cpp:304) null pointer passed for argument "device"
> [ WARN] [1601568480.178749109]: Device 1/1 failed with exception: failed to set power state
> [ERROR] [1601568480.178776222]: The requested device with  is NOT found. Will Try again.
> ```

Did you ever figure this one out?

Shunichi09, 2021-01-27:
In my case, when launching the container, using privileged option and  -v "/dev:/dev" volume option solved this problem.

Like

```shell
docker run --gpus all -d \
    --name container_name \
    -v "/dev:/dev" \
    --privileged \
    --rm \
    image \
    tail -f /dev/null
```

pringithub, 2021-01-31:
I needed to add the 99-* file as well for realsense to work.

Using Ubuntu 18.04 with ROS melodic. Seems this issue has been going on for months .. update the apt package?

billusbillus, 2021-04-28:
Hi, I'm facing the same error and adding 99-realsense-libusb.rules under /etc/udev/rules.d/ didn't work for me. 

I can see `uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435` when I run `dmesg`. 

I'm working on ubuntu 18.04 and installed with `apt-get install ros-melodic-realsense2-camera`. Also, Idk if it might be helpful, but I can't run `rs-enumerate-devices`. Got `command not found`. 

Could you help me ? Thanks for time !

yayashuxue, 2021-06-22:
> I needed to add the 99-* file as well for realsense to work.
> 
> Using Ubuntu 18.04 with ROS melodic. Seems this issue has been going on for months .. update the apt package?

Mine still fail to work after adding it... I am also using 18.04 with ROS melodic. Can you provide me the steps you take? Thanks!!

arikus1, 2022-03-21:
i had a same mistakes,
i use rpi 4 4gb , t265 and kermel version = 5.13.0-1017-raspi  , 
i just want to run rs-data-collect and its end like this
realsense error calling rs2_create_device(info_list:0xaaaaef733d50, index:0): unable to open device interface

i ruun a couple command another rs- and its still the same traouble that unable to open device intervace .
can some one help me pls what should i do 

im from asia n my english was so bad im sory  im a newbie with program

MartyG-RealSense, 2022-03-22:
Hi @arikus1  Your English is good, so don't worry.  :)

The T265 Tracking Camera does not have built in depth sensing and so it would not work with some of the example programs.    The link below has a list of the examples that can work with T265.

https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md#examples-and-tools

arikus1, 2022-03-23:
hi @MartyG-RealSense  thank u for reply my qs 

i try two of the phyton examples ,t265 basic and coordinates

first mistake come : module 'pyrealsense2' has no attribute 'pipeline'
i try " import pyrealsense2.pyrealsense2 as rs" n it solve mybe 
but there is another mistakes that "run time Error: No device connected"
i alredy connected the camera n i run the realsense viewere too.

arikus1, 2022-03-23:
hi @MartyG-RealSense thank u for helping me , i wish u all the best broo , ur answer is a miracle for newbie like me 🙏

MartyG-RealSense, 2022-03-23:
You are very welcome, @arikus1 - I am pleased that I could help  :)

goyalvarun02, 2022-08-26:
> @hamsterasu I reproduced the issue and found one solution. Could you please check if you have 99-realsense-libusb.rules under /etc/udev/rules.d/ ? If not, please copy [...]

Hi ,

I tried the above but still it failed.

I am using docker on VM (ubuntu 18.04) .

[image]
[image]

MartyG-RealSense, 2022-08-26:
Hi @goyalvarun02  Please create a new case for your problem using the green **New Issue** button on the front page of this support forum.  Thanks!

https://github.com/IntelRealSense/librealsense/issues

goyalvarun02, 2022-08-26:
[Realsense d435i is not detected on docker #10829 ](https://github.com/IntelRealSense/librealsense/issues/10829)

Created :-)

yang6746, 2024-03-13:
I copied https://github.com/IntelRealSense/librealsense/blob/master/config/99-realsense-libusb.rules to /etc/udev/rules.d/, however, I got another error:

 13/03 23:18:38,690 WARNING [140163864020736] (messenger-libusb.cpp:66) bulk_transfer returned error, endpoint: 0x1, error: Resource temporarily unavailable, err. num: 11