realsenseai/realsense-ros #1409: Getting RAW data from realsense.

aravindsairam, 2020-09-21:
Hello,

Is it possible to obtain the RAW data from the realsense. By RAW data, I mean the data without any pre-processing or post-processing. I know there is a JSON file which does the pre-processing of the data where we can set all the parameters. But what I need is the data before pre-processing of those parameters. I need this data as my team and I are facing this weird noise(ghosting). These ghosting are occurring due some repetitive patterns, glare or even with slight improper lighting conditions. This makes our robot confused and it takes a different path as it thinks that these ghosting are a real obstacle on it's path.

We want to check how this data looks before any pre-processing and fine tune each parameters that suits our working condition. I am new to this realsense environment. Sorry If I am too naive. 

We are using intel realsense D435 with python.

MartyG-RealSense, 2020-09-21:
Hi @aravindsairam   The RealSense SDK provides a **Low-Level API** for using callbacks to access sensor data at the camera hardware (typically, librealsense programs are processed by the High Level API).

https://github.com/IntelRealSense/librealsense/blob/master/doc/api_arch.md#low-level-device-api

I also found this information about low-level API callbacks:

-  Minimum SDK-imposed overhead == minimum latency.
-  Each frame is handled with a separate callback - “first come – first serve”.
-  All the synchronization stuff is left for the user side to implement.
-  Check **rs-data-collect** tool of the SDK that uses this method

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect 

Source: https://github.com/IntelRealSense/librealsense/issues/4185#issuecomment-501679687

There are however ways to deal with ghost noise and repetitive patterns.  The links below provide examples of some excellent references.

**Repetitive patterns**
https://github.com/IntelRealSense/librealsense/issues/6713

**Ghost noise**
https://github.com/IntelRealSense/librealsense/issues/4553
https://github.com/IntelRealSense/librealsense/issues/7021

MartyG-RealSense, 2020-09-29:
Hi @aravindsairam   Do you still require assistance with this problem, please?   Thanks!

aravindsairam, 2020-09-30:
Thank you @MartyG-RealSense. It was really helpful.

MartyG-RealSense, 2020-09-30:
Thanks @aravindsairam  I'm glad I could be of help.  Plese feel free to close this case if you are satisfied with the outcome and have no further questions about it.  Thanks!

closed 2020-10-08 by MartyG-RealSense