realsenseai/realsense-ros #1410: Recording .bag file with IMU data combined T265

andriskuciks, 2020-09-21:
Hello!
 I've been looking for a quick solution but couldn't find one. I'm trying to record test data which will be used in most popular open source SLAM solutions like ORB_SLAM2/3 VINS-FUSION and RTABMAP. I have a IntelRealsense T265 tracking module, and I tried to record the data using `RealsenseViewer` recorder. I notice that accelerometer and gyroscope are posting data on different topics, but I need that those data are under the same topic. I know that there is this `unite_imu_method` when running `realsense2_camera` but is it possible to set this flag somehow in RealsenseViewer? 
Or there is maybe another easy solution for my problem? 
I don't really want to write my own program, just want to record data with the available tools but it's starting to look that it is impossible, could it really be?

doronhi, 2020-09-24:
The `unite_imu_method` is not available in realsense-viewer nor anything like it.
The best way would be to run realsense2_camera and run `rosbag record` on a different terminal, specifying the topics you are interested in recording.
Realsenense-viewer bag files are a little different in format from the regular rosbag files so you would have had problems using them with other open-source SLAM solutions anyhow.

andriskuciks, 2020-09-24:
Okay, thanks for your input!

closed 2020-09-24 by andriskuciks