realsenseai/realsense-ros #1411: How to get the (3D) position from d435i (and possibly with L515)?

mdasari823, 2020-09-21:
Hi,

I am using d435i for SLAM according to these instructions: https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i. I am able to run it successfully and see the depth maps up to 3m. I was wondering if there is a support to get the 3D translational pose directly from this. I couldn't find any as such.

I was going through this [issue](https://github.com/IntelRealSense/realsense-ros/wiki/SLAM-with-D435i), @MartyG-RealSense pointed out few directions. I just wanted to check with you if this repo has direct support before trying out other options.

RealSenseSupport, 2020-09-25:
Hi @mdasari823 

Let me see if I can help answer some of your questions but there are some confusion to the links that you have embedded in your description.  The "issue" link that you provided is the same link as the SLAM instruction page.  Do let me know what "issue" you were referencing.  So not sure what repo you are referencing as to supported or not.  Do let me know what repo you are referring to so that we can provide you with the proper support effort.

The translation can be obtained by running, in a separate terminal, the following command:
rosrun tf tf_echo /map /camera_link

Are you asking if position is in ROS topic and not just in static-transform format?  You may be able to see it running here:

/rtabmap/odom
rostopic echo /rtabmap/odom/pose/pose

mdasari823, 2020-09-28:
@RealSenseCustomerSupport, ah, my bad I updated the proper link that I mentioned. Please check. And thanks for suggesting the commands to get the pose. I am getting the pose but it (the RGB-D SLAM) seems it is really unstable for hand motion. It is okay when I use trolly for moving the camera.  I was wondering if this is a code problem or the limitation of the  SLAM algorithm implemented in this repo.

doronhi, 2020-09-29:
I think what you describe here is the normal behavior for this example. As I already mentioned [here](https://github.com/IntelRealSense/realsense-ros/issues/951#issuecomment-542112046), I believe that with better understanding of the packages involved in this process better results are achievable.

mdasari823, 2020-09-29:
Thanks @doronhi, so normal behavior as in that is because of the SLAM algorithm or lack of enough data from the depth sensors? My question is basically, why T265's tracking is really good compared to these depth sensors.

doronhi, 2020-09-29:
I am not familiar with either the specific SLAM algorithm in the T265 nor in the rtabmap package so I can't tell if that's the source of the difference in performance.
I also don't know the quality difference between the IMUs in the T265 and the D435i.
The T265 has a huge FOV compared to the D435i and that's very helpful.
There are more differences stemming from T265 making all the calculation onboard and basically built for SLAM.
But for me, at the time, it seemed that the d435i's IMU data simply wasn't really taken into account by the rtabmap and that's what damaged the stability the most. Therefor I assumed that the parameters in the "opensource_tracking.launch" file were wrong.
The way to prove that that's the issue is to find the right configuration but I never got around to investigating enough. I do hope to get a confirmation and a correction to the demo.

mdasari823, 2020-09-29:
Yeah, the FoV makes sense but the lack of IMU correction definitely makes more sense for the stability issues. I am using a launch that was suggested [here](https://github.com/introlab/rtabmap/issues/574#issuecomment-689149729), uses the IMU for the L515 Lidar sensor. But it seems the stability issue still exists (for both RGB-D & ICP odometry).

blakely3dnm, 2020-10-01:
I'd recommend running the realsense IMU calibration script of you intend to use the IMU for SLAM.

mdasari823, 2020-10-01:
So I am using IMU now. I think I didn't mention this before. The following is the problem I am facing when I say stability issues. 

```
[ WARN] (2020-10-01 12:35:10.950) OdometryF2M.cpp:533::computeTransform() Registration failed: "RegistrationIcp cannot do registration with a null guess."
[ERROR] (2020-10-01 12:35:10.986) RegistrationIcp.cpp:1301::computeTransformationImpl() RegistrationIcp cannot do registration with a null guess.
[ WARN] (2020-10-01 12:35:10.986) OdometryF2M.cpp:533::computeTransform() Registration failed: "RegistrationIcp cannot do registration with a null guess."
[ WARN] [1601570110.991484422]: Could not get transform from odom to l515_link after 0.200000 seconds (for stamp=1601570110.480764)! Error="Lookup would require extrapolation into the future.  Requested time 1601570110.480764289 but the latest data is at time 1601570018.199666467, when looking up transform from frame [l515_link] to frame [odom]. canTransform returned after 0.20018 timeout was 0.2.".
[ERROR] (2020-10-01 12:35:11.022) RegistrationIcp.cpp:1301::computeTransformationImpl() RegistrationIcp cannot do registration with a null guess.
[ WARN] (2020-10-01 12:35:11.022) OdometryF2M.cpp:533::computeTransform() Registration failed: "RegistrationIcp cannot do registration with a null guess."
[ERROR] (2020-10-01 12:35:11.059) RegistrationIcp.cpp:1301::computeTransformationImpl() RegistrationIcp cannot do registration with a null guess.
[ WARN] (2020-10-01 12:35:11.059) OdometryF2M.cpp:533::computeTransform() Registration failed: "RegistrationIcp cannot do registration with a null guess."
```

Do you have any idea? I think there should be a way of correcting/resetting the odometry when it has a null guess.

pranav-pang, 2021-04-06:
How can I find what type of messages /rtabmap/odom/pose/pose contains? If I wanted to write a subscriber to this topic so that I can plot the pose over time?

RealSenseSupport, 2021-08-30:
@mdasari823 Do you need further help? If not, this issue will be closed in 7 days.