realsenseai/realsense-ros #1412: Use threshold filter

harshmittal2210, 2020-09-22:
Hy, 
I am not able to find a way to add a threshold filter in the launch file. Can you let me know a way to add it as I am not able to find a way in documentation as well. Using realsense-viewer I am to see the result using the threshold filter in post processing. How can I use the same in ros?

MartyG-RealSense, 2020-09-22:
Hi @harshmittal2210   Doronhi the RealSense ROS wrapper developer has stated that the threshold filter is not in the RealSense ROS wrapper but you can instead set a maximum distance with the **clip_distance** flag.

https://github.com/IntelRealSense/realsense-ros/issues/413

The RealSense ROS wrapper's front-page documentation provides this description for clip_distance: "remove from the depth image all values above a given value (meters). Disable by giving negative value (default)".

harshmittal2210, 2020-09-25:
Okay, Thanks

closed 2020-09-25 by harshmittal2210