realsenseai/realsense-ros #1413: Specify stream type for IR imagers on D415

mjsobrep, 2020-09-22:
Occasionally my IR imagers stream in color, but most of the time are streaming grayscale (visualizing in Rviz). I don't see a launch parameter or dynamic reconfig parameter in ros to control the streaming format. How do we specify that?

MartyG-RealSense, 2020-09-23:
Hi @mjsobrep  I would think that it would be a matter of which camera topic you subscribe to in order to get color or grayscale infrared:

**Color:**  /camera/color/image_raw
**IR 1:**  /camera/infra1/image_rect_raw
**IR 2:**  /camera/infra2/image_rect_raw

You can enable and disable stream types in the roslaunch statement.

**Infra**
enable_infra1:= false
enable_infra1:= true

enable_infra2:= false
enable_infra2:= true

**Color**
enable_color:=false
enable_color:=true

To change the setting inside the roslaunch file instead of the roslaunch statement:

**Infra**
<arg name="enable_infra1" default="false" />
<arg name="enable_infra1" default="true" />

<arg name="enable_infra2" default="false" />
<arg name="enable_infra2" default="true" />

**Color**
<arg name="enable_color" default="false" />
<arg name="enable_color" default="true" />

Each dynamic_reconfigure variable has an equivalent in rosparam with the same name.  So the **enable_** variable names should be the same for a rosrun rqt-reconfigure instruction as they are for the launch file rosparams.

mjsobrep, 2020-09-23:
Hey @MartyG-RealSense , thanks. This is a bit different than what I am looking for. The D415 stereo imagers are color. But they can also stream in grayscale. I'm looking to control that. 

In realsense-viewer this is called "available streams", I need to select rgb or bgr
[image]

MartyG-RealSense, 2020-09-23:
I had considered that you might also be referring to color from the IR imager, so I researched that possibility at the same time as assembling the information that I posted earlier.

The easiest approach may be to turn the IR Emitter off to remove the IR dot pattern, giving you an RGB-like grayscale image.  You can turn the emitter off with a rosparam in the launch file or toggle it on / off with dynamic_reconfigure during runtime.

https://github.com/IntelRealSense/realsense-ros/issues/1379#issuecomment-691453842

mjsobrep, 2020-09-23:
Since I am using a D415, I should be able to get color with the IR emitter on. This is described in 10(c) in the [tuning guide](https://www.intelrealsense.com/wp-content/uploads/2019/11/BKMs_Tuning_RealSense_D4xx_Cam.pdf). Note, I can see get color out of the left IR camera in the realsene viewer (by setting that stream value to rgb or bgr), I just can't figure out how to get it in ROS.

MartyG-RealSense, 2020-09-23:
Not all functions supported in the librealsense SDK are available in the RealSense ROS wrapper too.

I will tag the ROS wrapper developer @doronhi to seek his input on the subject.

mjsobrep, 2020-09-23:
Ok, thanks.

mjsobrep, 2020-09-25:
@MartyG-RealSense Did you hear back anything?

MartyG-RealSense, 2020-09-25:
No I didn't.  I will tag a reminder to @doronhi

mjsobrep, 2020-09-25:
Ok, thanks. 

I did a bit of digging in the code. Looks like there is some [hard coded stuff for the infrared stream format](https://github.com/IntelRealSense/realsense-ros/blob/de64e737c6a9de4ec70d89a18fe68b526c1460f5/realsense2_camera/src/base_realsense_node.cpp#L96). I don't know if just changing those lines to would be sufficient to change to color, maybe allowing the user to choose based on a launch file flag. Any insight that @doronhi would definitely be great.

doronhi, 2020-09-27:
Currently, as you know, IR sensors are hard coded to publish IR data formatted as gray scale images. This is reflects well the capabilities of most devices.
I expect that just changing those lines will not be enough, that the idea that IR images are gray-scale images is very much inherent in the code. On the other hand, I also expect that enough playing around with the code will get you the color image that you want.
Provided with your use case, we can consider a change request. If you try it yourself, your experience may be very beneficial to this discussion.

mjsobrep, 2020-09-27:
The use case is to be able to have color images that are well aligned in both time and space to the depth map. Using the left color+IR imager on the D415 is the best way to do that according to the Intel tuning guide (which makes a lot of sense to me). Being able to access that stream in color is one of the benefits to us of the D415. 

I will see if I can play around with the code a little bit later this week and post back what I find. If you have some time to take a look as well, I would really appreciate it.

doronhi, 2020-09-27:
It's sounds reasonable enough and I'm quite surprised it didn't come up more often, if it came up at all, until now.
Unfortunately, I don't have a D415 available right now so I won't be able to make real contribution for a while.
I am working in the background on a new version with a more flexible design. Right now, of the top of my head, I think I will put a note to integrate that capability there, rather then spend the time upsetting the current code. As I can't give any timeline yet, I'll understand if you decide to try adding that option into the current version. In that case I'll be happy to consider a PR.

MartyG-RealSense, 2020-10-05:
Hi @mjsobrep   Do you still require assistance with this case, please?   Thanks!

mjsobrep, 2020-10-05:
I think this should stay open

MartyG-RealSense, 2020-10-05:
@mjsobrep   As mentioned above, @doronhi  made a note to consider creating the function at a future date.  Unless you want to attempt to create a PR to add the feature yourself then there may not be much to be gained from keeping this case open, since @doronhi has already taken note of the issue.  Thank you.

mjsobrep, 2020-10-06:
yup, I'm going to try to get a PR put together when I have a chance

MartyG-RealSense, 2020-10-06:
Okay thank you, I will leave this case open for a further period then.  Please do keep us updated about the progress of your PR project.   Thanks!

MartyG-RealSense, 2020-10-17:
Adding a note to keep this case open for a further period.

doronhi, 2020-11-04:
This option was just added in #1488 . Please check if it meets your needs.

mjsobrep, 2020-11-10:
@doronhi #1488 resolves this request. I have tested it with two D435 cameras running. Seems to work well.

mjsobrep, 2020-11-17:
#1488 is merged, so this can be closed.

closed 2020-11-17 by mjsobrep

MartyG-RealSense, 2020-11-17:
@mjsobrep   Thanks for the update!