realsenseai/realsense-ros #1414: ROS subscribe multiple camera topics can not go into callback function

YunchuZhang, 2020-09-22:
Hi there,
I tried to subscribe 2 cameras' topic and show them together(camera1 's rgb, cameara2's depth).But it can not go into callback function. If I change to subscribe same camera's topic and show them, then it works. Could you help me solve this? Here is my code, I use TimeSynchronizer to make them filtered : 

#!/usr/bin/env python

import rospy
import message_filters
from sensor_msgs.msg import Image, CameraInfo
from cv_bridge import CvBridge, CvBridgeError
import cv2
import numpy as np

class LoadImage(object):

    def __init__(self):
        self.rcgb1_image = None
        self.depth1_image = None
        self.rgb1_image_sub = message_filters.Subscriber("/camera1/color/image_raw",Image)
        self.depth1_image_sub = message_filters.Subscriber("/camera2/aligned_depth_to_color/image_raw",Image) 
        self.test = message_filters.Subscriber("/camera1/depth/camera_info",CameraInfo) 

        self.bridge_object = CvBridge()
        self.ts = message_filters.TimeSynchronizer([self.rgb1_image_sub, self.depth1_image_sub, self.test], 100)
        self.ts.registerCallback(self.fusion)
    def fusion(self, rgb1, depth1, test):
        rospy.loginfo(test)
        rospy.loginfo("This message will print only once")

        try:
            # We select bgr8 because its the OpenCV encoding by default
            self.rgb1_image = self.bridge_object.imgmsg_to_cv2(rgb1, desired_encoding="bgr8")
        except CvBridgeError as e:
            print(e)
        try:
            # We select 32FC1 because its the OpenCV encoding by default
            self.depth1_image = self.bridge_object.imgmsg_to_cv2(depth1, desired_encoding="32FC1")
        except CvBridgeError as e:
            print(e)
        if self.depth1_image is not None and self.rgb1_image is not None :

            cv2.imshow('rgb1_image',self.rgb1_image)
            # cv2.waitKey(1)
            cv2.imshow('depth1_image',self.depth1_image)
            cv2.waitKey(1)

def main():
    rospy.init_node('load_image_node', anonymous=True)
    rospy.loginfo("Starting load_image_node node")
    load_image_object = LoadImage()
    rospy.sleep(.1)
    try:
        rospy.spin()
    except KeyboardInterrupt:
        print("Shutting down")
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()

MartyG-RealSense, 2020-09-23:
I should emphasize upfront that this topic is a bit outside of my knowledge but I will do my best to assist.

As a starting point, the discussion in the link below seems to be similar to what you are trying to achieve.

https://stackoverflow.com/questions/57119292/how-to-merge-2-ros-topics-into-1

doronhi, 2020-09-23:
I am also not familiar with "message_filters" but it could be related to different time domain. When running 2 cameras, try `echo /camera1/color/image_raw/header` and `echo /camera2/aligned_depth_to_color/image_raw/header` (from memory) on 2 side-by-side terminals and check if the time stamps seem similar enough.
Also, validate that global-timestamp domain is available by running realsense-viewer with both cameras.

YunchuZhang, 2020-09-23:
Thanks MartyG, I do take. reference from that. But it comes the error like I shown

YunchuZhang, 2020-09-23:
Hi Doronhi, the time stamps are similar, but still can not run through the callback smoothly.  
[image]
Btw, where should I set the global-timestamp?
[image]

doronhi, 2020-09-24:
Assuming the control "Global Time Enabled", you should see the "Timestamp: Global Time" tag when hovering over the timestamp in the image:
[image]
If it seems ok I think we should try and stop guessing why "message_filters" does not pass on the frames and search for the answer in it's logs or code. We may be wrong to assume it the time stamps...
Once you know why the frames are not being passed on, then you should try to find out what calibration is needed from the cameras.

YunchuZhang, 2020-09-24:
I check the global time enabled thing, it was correct. 
The time stamps seems correct

closed 2020-10-12 by MartyG-RealSense