realsenseai/realsense-ros #1417: Frequency diagnostics not published until getNumSubscribers > 0

mikepurvis, 2020-09-24:
Do to issues with hardware and firmware reliability, our robot has a power cycle retry loop on our realsense cameras. We use the built-in FrequencyDiagnostics from this driver to determine whether the camera is up, to avoid our monitoring process needing to actually receive all of the published image data.

Unfortunately, the way the driver is structured at present, the frequency diagnostic does not get updated or even published unless the RealSense image topics are being subscribed to. The diagnostic update happens in `publishFrame`, here:

https://github.com/IntelRealSense/realsense-ros/blob/e273f04d480f9331b1919f24f59e7712fd2bc688/realsense2_camera/src/base_realsense_node.cpp#L2233

And there are several places this is guarded behind a `getNumSubscribers > 0` condition, both one internal to `publishFrame`, but also a separate one in `publishAlignedDepthToOthers`, which does additional processing before calling through to `publishFrame`.

In my opinion, this is counter to the spirit of diagnostics, and could be especially confusing in cases such as a gripper cam where the logic which consumes the images is intentionally shut down when the gripper is not actively in use. Under the current model, that device would get flagged by diagnostic_aggregator as having a stale diagnostic during the time that there is no active subscriber.

Would we entertain a pull request to fix this so that the frequency diagnostics are always published regardless of subscriber status? Currently the calls are:

```
frame_callback -> publishAlignedDepthToOthers -> publishFrame -> publish
frame_callback -> publishFrame -> publish
```

IMO the non-depth case should be moved to a utility function similar to `publishAlignedDepthToOthers`, and that would make all of this logic easier to follow and understand. However, for the very immediate, my proposed approach would be to move the subscriber check from `publishFrame`, making it instead external to where it is called from `frame_callback`, and then outside that check (in both the depth and non-depth cases), calling the `FrequencyDiagnostics::update` method.

Does this approach seem reasonable? I also note that the current ROS 2 branch of this repo doesn't have any of the diagnostics stuff included at all, so I'm not sure how strong the commitment is to it.

doronhi, 2020-09-25:
Hi @mikepurvis ,
Thank you for this elaborate use case.
I agree with what you said and would love to have a PR according to your suggestion.

As for the ROS2, the topic is currently missing only because of a technical issue:
I used "diagnostic_updater::HeaderlessTopicDiagnostic" but in it's destructor it also destroys my node. This happens when the camera is disconnected. I thought of moving the entire object to the "RealSenseNodeFactory" class but it feels like there is a better way and I didn't get around to it yet. This is definitely on my list, especially now with your comment.
You are more then welcome to check it out [here](https://github.com/doronhi/realsense/tree/eloquent-diagnostics). Let me know if there are permission issues.

RealSenseSupport, 2020-10-05:
Hi @mikepurvis Did you submit the PR? Do you need further help on this issue? If we don't hear from you in 7 days, this issue will be closed.

rgariepy, 2020-10-06:
@RealSenseSupport This issue still exists, and no, we haven't done the work to fix someone else's product yet.

RealSenseSupport, 2020-10-09:
@rgariepy Sorry I misunderstood the previous discussion(have thought you'll submit one PR). As doronhi's replied, could you please help check if https://github.com/doronhi/realsense/tree/eloquent-diagnostics works for you? Thanks!

doronhi, 2020-10-12:
> @rgariepy Sorry I misunderstood the previous discussion(have thought you'll submit one PR). As doronhi's replied, could you please help check if https://github.com/doronhi/realsense/tree/eloquent-diagnostics works for you? Thanks!

A clarification is needed, I feel:
The branch mentioned here https://github.com/doronhi/realsense/tree/eloquent-diagnostics isn't working yet. I suggested anyone with background using diagnostics on eloquent to have a look and help with this branch. In any case, I hope to get on with it in the next couple of days and have it working and merged in the main eloquent branch.

RealSenseSupport, 2020-10-12:
@dorodnic  Thanks for the clarification!  Looking forward to the update.

kubja, 2020-10-28:
Hi @mikepurvis,
I was facing similar issues when using the diagnostics for Intel Realsense to monitor the camera health a was running a fix. I just committed the changes and created a pull request #1471 if that was of interest. I went a step further and separated "update" from "tick" function so that the frequency message update is published even when no frames are coming.

mikepurvis, 2020-10-28:
Thanks for looking into this, @kubja. I'll close this ticket in favour of your PR.

closed 2020-10-28 by mikepurvis