realsenseai/realsense-ros #1418: Motion Module force pause

miajl, 2020-09-25:
I have a realsense 435i.

I used to be able to get imu messages from the camera running this:
`roslaunch realsense2_camera rs_camera.launch unite_imu_method:=linear_interpolation enable_infra1:=true enable_infra2:=true enable_gyro:=true enable_accel:=true`

But now I get this error and no longer get messages on the imu topic.

`[ WARN] [1601040637.128804926]: Hardware Notification:Motion Module force pause,1.60104e+12,Error,Hardware Error`

Reading online says that I need to do a camera calibration, but the camera was working fine before and the readme for the calibration script says 435i comes factory calibrated.

MartyG-RealSense, 2020-09-25:
Hi @miajl   I would first recommend trying adding the term **initial_reset:=true** to your roslaunch statement to perform a hardware reset of the camera.

miajl, 2020-09-28:
That worked thank you

closed 2020-09-28 by miajl

MartyG-RealSense, 2020-09-28:
Great news, thanks for the update!