realsenseai/realsense-ros #1420: D455 Connect a device

jinteahwa, 2020-09-29:
I installed it following the installation guide below.
(https://tks2.co.jp/2019/12/21/intel-realsense/)
The camera was caught in the viewer, but the Unity Project received the "Connect A RealSense Device" message, resulting in an error "ExternalException: rs2_pipeline_start_with_config".
What's the problem?

MartyG-RealSense, 2020-09-29:
Hi @jinteahwa  The issue of D455 not being detected in the RealSense Unity wrapper has already been reported to the RealSense team in the past week.  I have passed your own report to them.  Thanks for letting us know!

MartyG-RealSense, 2020-10-08:
Adding a note to keep this case open for a further time period whilst Intel investigations are ongoing.

MartyG-RealSense, 2020-11-29:
Hi @jinteahwa   I tested the Unity wrapper again with D455 and the new SDK **2.40.0** and it still did not work unfortunately.  I will ask Intel about it again.

doronhi, 2020-11-29:
I think this is an issue for the [librealsense2](https://github.com/IntelRealSense/librealsense) repository. Specifically, is it related to [#7662](https://github.com/IntelRealSense/librealsense/issues/7662)?

MartyG-RealSense, 2020-11-29:
Hi @doronhi   Yes, this is a librealsense issue.  Sometimes SDK issues get posted here instead of on the librealsense github.

doronhi, 2020-11-29:
I think we'll get better results if we posted that issue there. @jinteahwa, can you comment on [#7662](https://github.com/IntelRealSense/librealsense/issues/7662)? Or @MartyG-RealSense since you also managed to reproduce the issue.
As I am not a Unity user. Therefor I feel I should not be the one to promote this as a librealsense2 issue.

MartyG-RealSense, 2020-11-30:
@jinteahwa   Dorodnic the RealSense SDK Manager states in https://github.com/IntelRealSense/librealsense/issues/7662 that D455 is compatible with the 2017 versions of Unity, though I recognise that this is not ideal.

I have submitted a feature request for Unity 2019 / 2020 support for D455, linked to https://github.com/IntelRealSense/librealsense/issues/7883

MartyG-RealSense, 2020-12-14:
Case closed due to no further comments received.

closed 2020-12-14 by MartyG-RealSense

Sambhav-Viztech, 2020-12-15:
D455 is a nightmare

Alex-Beh, 2021-06-02:
@Sambhav-Viztech Can you elaborate more based on your experience?