realsenseai/realsense-ros #1422: /camera/color/image_raw topic not being published

cscholler, 2020-09-30:
I am trying to run SLAM with a D430. I have been trying to use the ORB_SLAM2 package which is designed to work with the RealSense line and subscribes to two ROS topics: an RGB image and a depth image. I have been trying to use the realsense-ros node to provide the those two topics for ORB_SLAM2, however, the real sense node only seems to be publishing the depth image (`/camera/depth/image_rect_raw`) but not the RGB image  (`/camera/color/image_raw`). As far as I can tell the camera seems to be running fine, I've been able to see both the infrared and depth stream using the real sense viewer. I have been running realsense with `roslaunch realsense2_camera rs_camera.launch initial_reset:=true` and the node appears to launch successfully, giving me the message "RealSense Node Is Up".

If it helps these are all the topics that the realsense node is publishing:
-     /camera/depth/camera_info 
-     /camera/depth/image_rect_raw
-     /camera/realsense2_camera_manager/bond 
-     /camera/stereo_module/auto_exposure_roi/parameter_descriptions 
-     /camera/stereo_module/auto_exposure_roi/parameter_updates 
-     /camera/stereo_module/parameter_descriptions 
-     /camera/stereo_module/parameter_updates 
-     /diagnostics 

I appreciate any suggestions you might have as to why the image topic isn't publishing.

Thank you!

MartyG-RealSense, 2020-09-30:
Hi @cscholler   The D430 Depth Module does not have an RGB sensor.  The only board in the RealSense Depth Module Kit range that has an integrated RGB sensor is the D415.  

Inside the casing of the **D435** cased USB camera that has a D430 module inside, there is an RGB sensor component that is separately attached via a cable to provide the RGB on that model.

cscholler, 2020-09-30:
That makes a lot of sense, thank you!

closed 2020-09-30 by cscholler

fangxiaomeng0830, 2023-03-31:
Hi, have you successfully run SLAM with D430I?

MartyG-RealSense, 2023-03-31:
Hi @fangxiaomeng0830  I have responded to your question about why your camera is mis-identified as D430i when it should be D435i.