realsenseai/realsense-ros #1423: Release for ROS Noetic

tylerjw, 2020-09-30:
Please cut a release for Noetic so this can be easily used on 20.04 with ROS1.

MartyG-RealSense, 2020-10-01:
Hi @tylerjw   I am not a RealSense ROS wrapper developer, so I do not have official information or comment to offer regarding any future plans regarding the wrapper and Noetic.

I have only seen one report about installing the current ROS1 wrapper for Ubuntu 20.04 under Noetic.  They compiled the wrapper from **source code** and reported that they were using it "without any specific issue"..

https://github.com/IntelRealSense/realsense-ros/issues/1270#issuecomment-655679810

doronhi, 2020-10-01:
Noetic version is planned, if everything will go according to plan, for December 2020.

MartyG-RealSense, 2020-10-08:
Case closed due to answer provided and no further comments received.

closed 2020-10-08 by MartyG-RealSense

DLu, 2020-10-16:
It would be nice if you left one of the Noetic issues open so that people could track its progress going through the rest of 2020.

MartyG-RealSense, 2020-10-16:
@DLu   I recommend keeping an eye on the ROS wrapper **Releases** page instead to see when Noetic is mentioned in release notes as new versions are published.

https://github.com/IntelRealSense/realsense-ros/releases