realsenseai/realsense-ros #1424: T265 ros node cannot subscribe image topic

robin-shaun, 2020-10-01:
Hello, after I modified `rs_t265.launch`, 
```
  <arg name="enable_fisheye1"     default="true"/>
  <arg name="enable_fisheye2"     default="true"/>

  <arg name="fisheye_fps"         default="5"/>
```
warning happened
```
[ WARN] [1601555869.190257854]: Given stream configuration is not supported by the device!  Stream: Fisheye, Stream Index: 1, Width: 848, Height: 800, FPS: 5
[ WARN] [1601555869.190349290]: Given stream configuration is not supported by the device!  Stream: Fisheye, Stream Index: 2, Width: 848, Height: 800, FPS: 5
```
I used USB3.1 and realsense-viewer can get images at 30 fps.

I installed the ros package in ubuntu16.04 using apt install

spanthr, 2020-10-19:
I feel you are missing the configuration in this way in the modified launch file. 
 <arg name="fisheye_width"       default="640"/>
  <arg name="fisheye_height"      default="480"/>
  <arg name="enable_fisheye"      default="true"/>

doronhi, 2020-10-20:
Using `rs-enumerate-devices` with a T265 connected, I can see the following supported profiles:
```
Stream Profiles supported by Tracking Module
 Supported modes:
    stream       resolution      fps       format   
    Fisheye 1	  848x800	@ 30Hz	   Y8
    Fisheye 2	  848x800	@ 30Hz	   Y8
    Gyro	 N/A		@ 200Hz	   MOTION_XYZ32F
    Accel	 N/A		@ 62Hz	   MOTION_XYZ32F
    Pose	 N/A		@ 200Hz	   6DOF
```

Therefor, the configuration given of 5 fps for the fisheye camera is not supported. Only 30 fps is supported.

closed 2026-10-23 by MartyG-RealSense