realsenseai/realsense-ros #1425: T265 with Kobuki, occupancy map and SLAM

rjam92, 2020-10-02:
Hello,

I have recently purchased a T265 and a D435 and despite demos of their collective use there is no guide that walks through how to:
1. Align the frames of the cameras with each other and a mobile robot/ROS frames
2. How to use the depth camera and T265 collectively 
3. Which topic to publish the wheel odometer data to for the T265.

Any help/guide in this regard will be appreciated

arjunsengupta98, 2020-10-04:
Hey @rjam92 

The occupancy mapping branch implements a 2D occupancy map formation with T265 and D435 :
https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping

rjam92, 2020-10-05:
thank you @arjunsengupta98 , 
While I figures most of the other questions out, I am not sure how to access the map that is created by the T265, since the whitepapers and the product details state that it solves the slam problem yet I can not see any way to build a map with the T265 alone.
Any help with this will be appreciated

arjunsengupta98, 2020-10-05:
@rjam92 

I had the same question in mind when I first started using the sensor. Turns out that the T265 builds and internal map using Intel's proprietary algorithm which they haven't made open source. This internal map can't really be visualised as such. The use of this internal map is for the robot/system to constantly know it's own position in the environment which is accomplished by subscribing to the odometery topic that the t265 publishes. 

It also helps solve the kidnapped robot problem using "static nodes" and exporting/importing the internal map using librealsense APIs. 

Check out these threads to have a better understanding- 

https://github.com/IntelRealSense/realsense-ros/issues/1155 

https://github.com/IntelRealSense/realsense-ros/issues/940

closed 2021-03-24 by RealSenseSupport

Sollimann, 2022-02-01:
Is there a paper documenting the steps in the **2D occupancy map D435+T265** implementation? I would like to read through the theoretical background for this implementation.