realsenseai/realsense-ros #1427: L515 Camera + IMU parameters

iamsavva, 2020-10-06:
Hi, 

For the L515, I am looking for the extrinsic between the RGB camera and the imu (translation + rotation matrices), as well as the information about the imu (noisy density and biases for the accelerometer and the gyroscope). Are these available? If so, where? I'm getting no information in the /camera/gyro/imu_info /camera/accel/imu_info topics. 

Many thanks in advance for your help. Cheers.

yrusobeautiful, 2020-10-21:
Hi,
Have you made any progress? I have a similar question.

doronhi, 2020-10-21:
While realsense2_camera node is running you can use `rosrun tf tf_echo camera_color_frame camera_gyro_frame` to obtain the transformation between the RGB camera and the imu. This yields the following:
```
- Translation: [-0.004, -0.013, 0.003]
- Rotation: in Quaternion [-0.003, 0.010, 0.008, 1.000]
            in RPY (radian) [-0.006, 0.020, 0.016]
            in RPY (degree) [-0.353, 1.157, 0.934]
```

Alternatively, without realsense2_camera running in the background, you can use `rs-enumerate-devices -c` which yields the following lines, among many others:
```
...
Extrinsic from "Color"	  To	  "Gyro" :
 Rotation Matrix:
   0.999846        -0.00616689       0.0164294     
   0.00649643       0.999777        -0.0200803     
  -0.0163019        0.0201839        0.999663      

 Translation Vector: -0.0129784867167473  0.00256138201802969  0.00410137791186571  
...
```
The difference in sign stems from using different coordinate systems.

The imu residing inside the L515 is of type [Bosch BMI085](https://www.bosch-sensortec.com/products/motion-sensors/imus/bmi085.html).
You can find details, including noise density and bias, in the linked spec sheet.

closed 2020-10-27 by iamsavva

zouyajing, 2021-05-13:
> While realsense2_camera node is running you can use `rosrun tf tf_echo camera_color_frame camera_gyro_frame` to obtain the transformation between the RGB camera and the imu. This yields the [...]

Thanks for your answer. I can read the extrinsic parameter of L515 now. But I am still confused about the difference due to coordinate system.
From REP-103 https://www.ros.org/reps/rep-0103.html, ROS use the order of forward-left-up for body representation. What's the order used in rs-enumerate-devices -c?

Can I just use the info from tf_echo if I use ROS to read the data of L515 ?
Best regards