realsenseai/realsense-ros #1428: D435i does not reach maximum exposure value

fiorano10, 2020-10-07:
The D435i don't seem to adjust to maximum exposure value under low light even if the auto exposure is on. The max I have seen it reach is 8500, when the max is 165000. Why is that?

MartyG-RealSense, 2020-10-07:
Hi @fiorano10   8500 is used as the default value for exposure in the RealSense Viewer.  Though it is possible to set the exposure **manually** to a maximum of **165000**, when auto-exposure is enabled I have never heard of environmental conditions that will cause exposure to reach anything near to that high value.

fiorano10, 2020-10-07:
Not saying it needs to reach 165000, but why does it not go over 8500? For eg, to properly expose the scene, I have to set it manually to 25000, shouldn’t auto-exposure be able to adjust to higher levels than just 8500?

fiorano10, 2020-10-07:
I’m just wondering if there is a way to set the auto-exposure max level to something higher so I don’t have to do it manually.

MartyG-RealSense, 2020-10-07:
What method do you use to monitor the current exposure value in ROS during runtime when auto-exposure is active, please?   I ask because the exposure metadata value does not update whilst auto-exposure is active.

fiorano10, 2020-10-07:
I can't see the value. I can estimate the value by setting it manually and comparing it when auto-exposure is on. What is the max value auto-exposure can set itself to? Is it 165000?

MartyG-RealSense, 2020-10-07:
The subject of a function for setting a maximum exposure time has been discussed before.  The most recent response from a member of the RealSense team about such a feature that I could find was in March 2020, where it was stated that it was in the pipeline but there was not yet a schedule for implementation.

https://github.com/IntelRealSense/librealsense/issues/3925

MartyG-RealSense, 2020-10-07:
Reviewing the latest features in SDK **2.39.0**, a new feature called **Depth HDR** may be of interest to you.

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#new-features

"A new Depth enhancement feature implemented in both Host and the Device Firmware that allows to improve Depth data in adverse light conditions by merging data from consecutive frames.  At its core the feature allows to configure and run Depth and IR stream with per-frame specified exposure and gain values (sequence of 2 frames).  Dedicated metadata attributes allow to associate the arriving frames with the configuration established by user".

fiorano10, 2020-10-07:
I think my question/request is too vague. Let me rephrase, what is the maximum value that auto_exposure can set when enabled? Is it 165000?

MartyG-RealSense, 2020-10-07:
Your question was not too vague, I understood your point.  :)  For manual exposure it is 165000.  For auto-exposure, it is not clear if a maximum exists, and https://github.com/IntelRealSense/realsense-ros/issues/1428#issuecomment-705034958 above suggests that it does not at this current time but is on the developers' radar.

MartyG-RealSense, 2020-10-14:
Hi @fiorano10   Do you require further assistance with this case, please?  Thanks!

fiorano10, 2020-10-17:
Nope, we will set the exposure manually when needed. Thanks!

closed 2020-10-17 by fiorano10