realsenseai/realsense-ros #1431: dynamic reconfigure update does not work

fiorano10, 2020-10-08:
I have the following packages were upgraded, and I can no longer control dynamic parameters of the camera. It lists them, but can't set them
```
ros-melodic-dynamic-reconfigure/bionic,now 1.6.3-1bionic.20200820.230543 
ros-melodic-ddynamic-reconfigure/bionic,now 0.2.0-0bionic.20200820.232903]

```

MartyG-RealSense, 2020-10-08:
Hi @fiorano10  

Have you tried rebuilding the RealSense ROS wrapper after this update has occurred, please?

fiorano10, 2020-10-08:
I have tried doing `catkin clean` and then rebuilding. Also tried deleting devel and build folders and re-build. It just gets stuck on `Waiting to configure`

MartyG-RealSense, 2020-10-08:
Although kernel 5.4 is not officially validated for librealsense yet, it has been in use successfully for some months now with numerous different librealsense versions without many problems.  So it does not seem likely that the problem is related to an incompatibility with 2.38.1.  

It sounds as though you have gone back to SDK 2.33.1 but now find that dynamic reconfigure is no longer responding to settings changes.  Is that correct please?

If SDK 2.38.1 was installed with packages, can you also confirm please that you cleaned up 2.38.1 by following the package uninstall process in the link below:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md#uninstalling-the-packages

fiorano10, 2020-10-08:
I went back to 2.33.1 and cleaned up 2.38. I upgraded ddyanmic-reconfigure and dynamic-reconfigure but can't go back on those since only the latest version is available on apt

fiorano10, 2020-10-09:
Is there a specific version of dynamic reconfigure that it not supported? I need this to work to disable auto-exposure and set it manually

MartyG-RealSense, 2020-10-09:
I researched your problem again about dynamic reconfigure not changing after the packages below were updated:

ros-melodic-dynamic-reconfigure/bionic,now 1.6.3-1bionic.20200820.230543 
ros-melodic-ddynamic-reconfigure/bionic,now 0.2.0-0bionic.20200820.232903]

I was not able to find a solution to your questions though, so I will refer the issue to @doronhi the RealSense ROS wrapper developer.

doronhi, 2020-10-12:
Initially I had ros-melodic-dynamic-reconfigure version 1.6.0-0bionic.20191008.194626 installed and everything worked fine.
Then I upgraded to 1.6.3-1bionic.20200820.230543 and now when I launch "rqt_reconfigure" it doesn't even expend all the options. So I think there is some issue with ros-melodic-dynamic-reconfigure version 1.6.3. However, I am having trouble reverting it back to 1.6.0, other then reinstalling ros, to confirm.

fiorano10, 2020-10-12:
@doronhi, not sure how re-installing ROS would fix this, given it would just install the same version of dynamic-reconfigure. Unless reinstalling ROS upgrades some dependency which fixes this issue?

MartyG-RealSense, 2020-10-20:
Hi @fiorano10   Do you still require assistance with this case, please?   Thanks!

fiorano10, 2020-10-20:
Yes, how do I fix it?

MartyG-RealSense, 2020-10-20:
I have no information beyond what @doronhi said in https://github.com/IntelRealSense/realsense-ros/issues/1431#issuecomment-706882399 above, so I must defer to his specialist ROS knowledge as the wrapper developer in this case.

doronhi, 2020-10-21:
I was on my way to open an issue in [dynamic_reconfigure repository](https://github.com/ros/dynamic_reconfigure) when I checked again the problematic machine and now it works fine. Maybe I had it reset in the meanwhile, I don't know.
Bottom line is that although I can't help with the problem, I don't believe it is relate specifically to realsense2_camera node. Therefor, I think this issue can be closed.
If you can reproduce the error on a clean machine maybe an issue should be opened for dynamic_reconfigure repository.

YoshuaNava, 2020-10-21:
I'm also experiencing this issue with librealsense v2.35.2 and realsense-ros v2.2.11, firmware v5.12.6.0, setting the param `inter_cam_sync_mode`.

I've printed out the params that can be set for each sensor, and it find it weird that the params for the rgb_camera:

```
1 Option: brightness
2 Option: contrast
3 Option: exposure
4 Option: gain
5 Option: gamma
6 Option: hue
7 Option: saturation
8 Option: sharpness
19 Option: frames_queue_size
42 Option: inter_cam_sync_mode
19 Option: frames_queue_size
36 Option: filter_magnitude
42 Option: inter_cam_sync_mode
43 Option: stream_filter
44 Option: stream_format_filter
45 Option: stream_index_filter
19 Option: frames_queue_size
42 Option: inter_cam_sync_mode
43 Option: stream_filter
44 Option: stream_format_filter
45 Option: stream_index_filter
```

Include the inter_cam_sync_mode, which is not the case when I see the rgb camera controls in realsense-viewer.

YoshuaNava, 2020-10-21:
I'm also experiencing this issue with librealsense v2.35.2 and realsense-ros v2.2.11, firmware v5.12.6.0, setting the param inter_cam_sync_mode.

@doronhi In the file `librealsense/src/core/options.h` in the method `supports_option) you search for valid options, for a given sensor. You do this based on a map key. I've found that for some reason the option `inter_cam_sync_mode` is always found in the valid options for my camera stereo_module, but due to the line `return it->second->is_enabled();` the mode is flagged as not supported, and I get an error when I try to set it to genlock, which reads `Exception thrown while processing service call: object doesn't support option #42`

MartyG-RealSense, 2020-10-30:
@YoshuaNava   Glad to hear that you found a solution for your problem.

At the earlier suggestion of @doronhi , I will now close this issue.

closed 2020-10-30 by MartyG-RealSense

doronhi, 2020-11-01:
Thank @YoshuaNava for your insight. Your correction to the ddynrec callback helps to illustrate the issue. I do believe that a more complete solution is needed for the problem of toggling sensors on and off so I hope in the near future you wont be needing your patch anymore.

LanceChen, 2020-11-13:
I'm experiencing the same issue as @YoshuaNava (setting the inter cam sync mode via dynamic reconfigure). The versions of software I am using:

- dynamic reconfigure 1.6.3
- librealsense 2.39.0
- realsense-ros 2.2.18

The interesting thing is I have two cameras; on the newer firmware 05.12.08.200 dynamic reconfigure of the sync mode does not work and produces the above error, but it succeeds on an older firmware 05.11.06.250

This is when I am setting the older master slave modes, not the new genlock modes.