realsenseai/realsense-ros #1432: Can I use "catkin build" instead of "catkin_make", and how can I fix this error?

ketaro-m, 2020-10-08:
I use Ubuntu 18.04.5 and ROS melodic 1.14.9

The installation instructs us to use "catkin_make" but I got the same issue as [#1397](https://github.com/IntelRealSense/realsense-ros/issues/1397). So I tried to use "catkin build" instead. I ran the following commands 
```
mkdir -p (my_working_directory)/catkin_ws/src  
cd (my_working_directory)//catkin_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^\d+\.\d+\.\d+" | tail -1`
cd ../../
catkin init
catkin clean
catkin config --cmake-args -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin config --install
```
and then
```
catkin build
```

However, I got these warnings saying "unsupported package build type: 'ament_cmake'" and building the packages such as "realsense2_camera" were skipped.

[image]

How can I fix this error and build these packages?

doronhi, 2020-10-08:
I'll have to fix the installation instructions. When checking out the last released version you have got the eloquent version. You need to checkout the last ROS1 version. Use the following checkout command instead of the existing one:
```
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
```
You should get version 2.2.17.
Let us know how it goes with the `catkin build` command.

ketaro-m, 2020-10-08:
Thank you for your reply! I think `catkin build` command successfully goes and 'catkin_tools_prebuild', 'realsense2_camera', and 'realsense2_description' are build.

However, when I tried to start the camera node typing
```
source devel/setup.bash
roslaunch realsense2_camera rs_camera.launch
```
it fails with the following errors.

> RLException: [rs_camera.launch] is neither a launch file in package [realsense2_camera] nor is [realsense2_camera] a launch file name
The traceback for the exception was written to the log file

I also tried the following command instead
```
roslaunch src/realsense-ros/realsense2_camera/launch/rs_camera.launch
```
but got these errors.

> Resource not found: realsense2_camera
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

I couldn't find `realsense2_camera` by tap completion after `roslaunch`. Is this a problem with "devel/setup.bash"?

doronhi, 2020-10-08:
I don't know the answer. First thing I'd do is try from a different terminal.
Then, there may be differences in the installation process between `catkin build` and `catkin_make` which was tested with this version. If you cannot use catkin_make I would like to know the reason and maybe we'll get convinced that instructions and modification should be made in order to make `catkin build` an option.

ketaro-m, 2020-10-09:
I already have ~/catkin_ws which was build with `catkin build` so I cannot use `catkin_make`. 
Plus, I tried to rename it temporary as ~/catkin_ws_tmp and create a new ~/catkin_ws for this and followed the instruction. But when I did 

```
catkin_make clean
```
The following error occurred.

> -- ~~  traversing 3 packages in topological order:
-- ~~  - realsense_camera_msgs (plain cmake)
-- ~~  - realsense2_camera (plain cmake)
-- ~~  - realsense2_node (plain cmake)
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/melodic/share/catkin/cmake/catkin_workspace.cmake:100 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:69 (catkin_workspace)
-- Configuring incomplete, errors occurred!
See also "/home/mech-user/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/mech-user/catkin_ws/build/CMakeFiles/CMakeError.log".

I attach the contents of the log files just in case [here](https://gist.github.com/ketaro-m/36f01be0b344e8da75a86ba8fcc7fef2)

ketaro-m, 2020-10-09:
Answering my previous error not finding realsense2_camera package after building with `catkin build` command, I could solve it if I skip `catkin config --install`.
Then the whole processes are as follows. RealSense node was successfully build up.

```
mkdir -p (my_working_directory)/catkin_ws/src  
cd (my_working_directory)//catkin_ws/src/
git clone https://github.com/IntelRealSense/realsense-ros.git
cd realsense-ros/
git checkout `git tag | sort -V | grep -P "^2.\d+\.\d+" | tail -1`
cd ../../
catkin init
catkin clean
catkin config --cmake-args -DCATKIN_ENABLE_TESTING=False -DCMAKE_BUILD_TYPE=Release
catkin build
source devel/setup.bash
roslaunch realsense2_camera rs_camera.launch
```

Anyway, if I insert  `catkin config --install` just before `catkin build`, install folder was made. If I source install/setup.bash instead of devel/setup.bash in this case, roslaunch succeeded.
Which is better, and could you give me a clear explanation if possible?

doronhi, 2020-10-10:
I'm not sure I'm the right person. That is a pure ROS question and I think you'll get much better answers from people in [ros.org](https://answers.ros.org/questions/)

ketaro-m, 2020-10-10:
I'd do that.
But thank you anyway! I could install this package successfully!

closed 2020-10-10 by ketaro-m