realsenseai/realsense-ros #1435: SLAM with T265 and D455 together?

Andyshen555, 2020-10-09:
I have been using D435i for slam and recently purchased D455. I knew that T265 would improve tracking, so if I want to use T265 and D455 together, how should I configure the launch file so that it uses odom from T265? Is it done already in current launch files?

RealSenseCustomerSupport, 2020-10-13:
----------------------------------------------

Dear Andyshen555
“**_Thank you for highlighting the [Issue/Feature]. We have moved our focus to our next generation of products and consequently, we will not be addressing this [Issue/Feature] in the T265.”_**

Andyshen555, 2020-10-13:
In this case, what is your production plan? Obviously IMU in D455 is still not good enough. When will the next T camera be launched?

RealSenseCustomerSupport, 2020-10-16:
----------------------------------------------

dear @Andyshen555,
unfortunately, there is no plan for next T camera will be launch

Andyshen555, 2020-10-16:
Wait, T265 is launched within a year and you have no plan to launch a next T series camera. In this case what does the "next generation of products" mean? Why is T265 not supported anymore?

RealSenseCustomerSupport, 2020-10-21:
----------------------------------------------

Dear @Andyshen555,
We cannot comment on product that hasn't been released.  We will only comment on products that have been released. "
We already have new product such as  L515 and D455 which feature IMU launched in the market but your will need to implement your own SLAM algo on your host

closed 2020-11-03 by RealSenseCustomerSupport