realsenseai/realsense-ros #1436: D435i ros on Raspberry4B

HeadReaper-hc, 2020-10-11:
My platform is Raspberry4B with Ubuntu Mate 20.04, and ROS version is neotic.
After installing librealsense and realsense-ros, I run the command `roslaunch realsense2_camera rs_camera.launch` and every thing is ok, but when I turn on the `enable_gyro` and `enable_accel`, the following error is displayed, is there any way to solve this problem?

... logging to /home/hc/.ros/log/2e604e3a-0b64-11eb-b152-819f8ba77156/roslaunch-hc-3945.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://hc:35431/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: True
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: True
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: False
 * /camera/realsense2_camera/filters: 
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
 * /camera/realsense2_camera/infra2_frame_id: camera_infra2_frame
 * /camera/realsense2_camera/infra2_optical_frame_id: camera_infra2_opt...
[... 68 lines cut ...]
[ INFO] [1602381051.485210733]: setupPublishers...
[ INFO] [1602381051.492490808]: Expected frequency for depth = 30.00000
[ INFO] [1602381051.607569493]: Expected frequency for color = 30.00000
[ INFO] [1602381051.690743585]: Start publisher IMU
[ INFO] [1602381051.694641252]: setupStreams...
[ INFO] [1602381051.697384789]: insert Depth to Stereo Module
[ INFO] [1602381051.697574289]: insert Color to RGB Camera
[ INFO] [1602381051.697701530]: insert Gyro to Motion Module
[ INFO] [1602381051.697783882]: insert Accel to Motion Module
 11/10 09:50:51,775 WARNING [281472711278960] (ds5-motion.cpp:480) IMU Calibration is not available, default intrinsic and extrinsic will be used.
 11/10 09:50:51,799 WARNING [281468949500272] (backend-hid.h:48) write_fs_attribute with gyro_3d-dev1 failed. The specified path /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0007/HID-SENSOR-200076.2.auto/iio:device1/trigger/current_trigger is not valid
 _**_11/10 09:50:51,800 ERROR [281472711278960] (types.h:313) Failed to open scan_element /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0007/HID-SENSOR-200076.2.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied_**
 **11/10 09:50:51,801 ERROR [281472711278960] (backend-hid.cpp:969) Hid device is busy!**
**[ERROR] [1602381051.802102881]: An exception has been thrown: Failed to open scan_element /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0007/HID-SENSOR-200076.2.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied
terminate called after throwing an instance of 'rs2::backend_error'
  what():  Failed to open scan_element /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-2/2-2:1.5/0003:8086:0B3A.0007/HID-SENSOR-200076.2.auto/iio:device1/scan_elements/in_anglvel_x_en Last Error: Permission denied**_
[camera/realsense2_camera_manager-2] process has died [pid 3970, exit code -6, cmd /opt/ros/noetic/lib/nodelet/nodelet manager __name:=realsense2_camera_manager __log:=/home/hc/.ros/log/2e604e3a-0b64-11eb-b152-819f8ba77156/camera-realsense2_camera_manager-2.log].
log file: /home/hc/.ros/log/2e604e3a-0b64-11eb-b152-819f8ba77156/camera-realsense2_camera_manager-2*.log
[camera/realsense2_camera-3] process has finished cleanly
log file: /home/hc/.ros/log/2e604e3a-0b64-11eb-b152-819f8ba77156/camera-realsense2_camera-3*.log
^C[rosout-1] killing on exit
[master] killing on exit
^C^C^Cshutting down processing monitor...
... shutting down processing monitor complete
done

MartyG-RealSense, 2020-10-11:
Hi @HeadReaper-hc   A past case of the **https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017** with the RealSense ROS wrapper and D435i suggests checking the **udev rules** as a first course of action:

https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017

HeadReaper-hc, 2020-10-12:
> Hi @HeadReaper-hc In a past case of the **Failed to open scan_element** error with the RealSense ROS wrapper and D435i when using **roslaunch realsense2_camera rs_camera.launch**, a RealSense team member suggests checking the **udev rules**:
> 
> [IntelRealSense/librealsense#3242 (comment)](https://github.com/IntelRealSense/librealsense/issues/3242#issuecomment-464775017)
Thankyou for your reply, that's really helpful, my problem have been solved.

MartyG-RealSense, 2020-10-12:
Great news @HeadReaper-hc - thanks for the update!

MartyG-RealSense, 2020-10-20:
Closing case due to solution achieved and no further comments received.

closed 2020-10-20 by MartyG-RealSense