realsenseai/realsense-ros #1441: compile error in Ubuntu 18.04

slamhuangxin, 2020-10-12:
hi, i download the latest version. and compile in my woarkspace by the  "catkin_make".  the version of ros is melodic. But it report errors like this. Can you give me some advise? thanks.
/home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp: In lambda function:
/home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:62:51: error: call of overloaded ‘begin(std::array<std::__cxx11::basic_string<char>, 2>&)’ is ambiguous
         auto it = std::find_if(begin(list_of_fault), end(list_of_fault),
                                                   ^
In file included from /usr/include/boost/range/functions.hpp:18:0,
                 from /usr/include/boost/range/iterator_range_core.hpp:38,
                 from /usr/include/boost/lexical_cast.hpp:30,
                 from /opt/ros/melodic/include/ros/transport_hints.h:34,
                 from /opt/ros/melodic/include/ros/subscribe_options.h:33,
                 from /opt/ros/melodic/include/ros/node_handle.h:42,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /opt/ros/melodic/include/image_transport/publisher.h:38,
                 from /opt/ros/melodic/include/image_transport/image_transport.h:38,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:8,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/usr/include/boost/range/begin.hpp:97:55: note: candidate: typename boost::range_iterator<C>::type boost::range_adl_barrier::begin(T&) [with T = std::array<std::__cxx11::basic_string<char>, 2>; typename boost::range_iterator<C>::type = std::__cxx11::basic_string<char>*]
 inline BOOST_DEDUCED_TYPENAME range_iterator<T>::type begin( T& r )
                                                       ^~~~~
/usr/include/boost/range/begin.hpp:106:61: note: candidate: typename boost::range_iterator<const T>::type boost::range_adl_barrier::begin(const T&) [with T = std::array<std::__cxx11::basic_string<char>, 2>; typename boost::range_iterator<const T>::type = const std::__cxx11::basic_string<char>*]
 inline BOOST_DEDUCED_TYPENAME range_iterator<const T>::type begin( const T& r )
                                                             ^~~~~
In file included from /usr/include/c++/7/string:51:0,
                 from /usr/include/c++/7/stdexcept:39,
                 from /usr/include/c++/7/array:39,
                 from /usr/include/c++/7/tuple:39,
                 from /usr/include/c++/7/bits/unique_ptr.h:37,
                 from /usr/include/c++/7/memory:80,
                 from /usr/include/boost/config/no_tr1/memory.hpp:21,
                 from /usr/include/boost/get_pointer.hpp:14,
                 from /usr/include/boost/bind/mem_fn.hpp:25,
                 from /usr/include/boost/mem_fn.hpp:22,
                 from /usr/include/boost/bind/bind.hpp:26,
                 from /usr/include/boost/bind.hpp:22,
                 from /opt/ros/melodic/include/class_loader/class_loader.hpp:35,
                 from /opt/ros/melodic/include/pluginlib/./class_list_macros.hpp:40,
                 from /opt/ros/melodic/include/pluginlib/class_list_macros.h:35,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:6,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/usr/include/c++/7/bits/range_access.h:58:5: note: candidate: decltype (__cont.begin()) std::begin(const _Container&) [with _Container = std::array<std::__cxx11::basic_string<char>, 2>; decltype (__cont.begin()) = const std::__cxx11::basic_string<char>*]
     begin(const _Container& __cont) -> decltype(__cont.begin())
     ^~~~~
/usr/include/c++/7/bits/range_access.h:48:5: note: candidate: decltype (__cont.begin()) std::begin(_Container&) [with _Container = std::array<std::__cxx11::basic_string<char>, 2>; decltype (__cont.begin()) = std::__cxx11::basic_string<char>*]
     begin(_Container& __cont) -> decltype(__cont.begin())
     ^~~~~
/home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:62:71: error: call of overloaded ‘end(std::array<std::__cxx11::basic_string<char>, 2>&)’ is ambiguous
         auto it = std::find_if(begin(list_of_fault), end(list_of_fault),
                                                                       ^
In file included from /usr/include/boost/range/functions.hpp:19:0,
                 from /usr/include/boost/range/iterator_range_core.hpp:38,
                 from /usr/include/boost/lexical_cast.hpp:30,
                 from /opt/ros/melodic/include/ros/transport_hints.h:34,
                 from /opt/ros/melodic/include/ros/subscribe_options.h:33,
                 from /opt/ros/melodic/include/ros/node_handle.h:42,
                 from /opt/ros/melodic/include/ros/ros.h:45,
                 from /opt/ros/melodic/include/image_transport/publisher.h:38,
                 from /opt/ros/melodic/include/image_transport/image_transport.h:38,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/include/../include/realsense_node_factory.h:8,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/include/base_realsense_node.h:6,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/../include/t265_realsense_node.h:3,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/t265_realsense_node.cpp:1:
/usr/include/boost/range/end.hpp:91:55: note: candidate: typename boost::range_iterator<C>::type boost::range_adl_barrier::end(T&) [with T = std::array<std::__cxx11::basic_string<char>, 2>; typename boost::range_iterator<C>::type = std::__cxx11::basic_string<char>*]
 inline BOOST_DEDUCED_TYPENAME range_iterator<T>::type end( T& r )
                                                       ^~~
/usr/include/boost/range/end.hpp:100:61: note: candidate: typename boost::range_iterator<const T>::type boost::range_adl_barrier::end(const T&) [with T = std::array<std::__cxx11::basic_string<char>, 2>; typename boost::range_iterator<const T>::type = const std::__cxx11::basic_string<char>*]
 inline BOOST_DEDUCED_TYPENAME range_iterator<const T>::type end( const T& r )
[... 133 lines cut ...]
 static void rs2_project_color_pixel_to_depth_pixel(float to_pixel[2],
             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/opt/ros/melodic/include/librealsense2/rsutil.h:150:13: warning: ‘void rs2_fov(const rs2_intrinsics*, float*)’ defined but not used [-Wunused-function]
 static void rs2_fov(const struct rs2_intrinsics * intrin, float to_fov[2])
             ^~~~~~~
realsense-ros-development/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/build.make:110: recipe for target 'realsense-ros-development/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o' failed
make[2]: *** [realsense-ros-development/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/t265_realsense_node.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
In file included from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/../include/../include/realsense_node_factory.h:13:0,
                 from /home/huangxin/catkin_ws/src/realsense-ros-development/realsense-ros/realsense2_camera/src/../include/base_realsense_node.h:6,
[... cut ...]
make[2]: *** [realsense-ros-development/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/src/base_realsense_node.cpp.o] Error 1
CMakeFiles/Makefile2:6811: recipe for target 'realsense-ros-development/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all' failed
make[1]: *** [realsense-ros-development/realsense-ros/realsense2_camera/CMakeFiles/realsense2_camera.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed

RealSenseSupport, 2020-10-30:
Hi

Thank you for pointing this out. Wheeled odometry support for ROS was never added and at this time we have moved our focus to our next generation of products and consequentially will not be addressing any new T265 features.

Thanks

RealSenseSupport, 2020-11-04:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

FPSychotic, 2020-11-14:
how is possible that if Intel advert wheel input is needed to get a reliable odometry, and  agree, you don't give the support? when it is not only necessary, it was advised will be added, I have the calibration file and it is linked to my ros launch, are you telling me it is not working, I followed the intel instructions that were given? 
I expent money in motors with encoders to use it with t265, I could just add any motor without them and save the money.
Did you abandoned either get a fix to the t265 be recognized at boot, and don't need plug and unplug every time? 2 years and it still is no fixed. 
It sounds for me to programmed obsolescence which is illegal in Europe

sasilva1998, 2020-12-01:
Agree with the two comments above, if support for ROS is advertised then how come are you gonna say that you will be no longer updating it, if no continuous update is given your device just becomes obsolete for what it is advertised.

closed 2026-10-23 by MartyG-RealSense