realsenseai/realsense-ros #1442: Depth cutoff adjust in SLAM?

Andyshen555, 2020-10-12:
It seems there is a depth cutoff when running opensource_tracking.launch. I wonder how we should adjust that, since rviz / rtabmap is using more info to match its visual odom than what we see in the point cloud.

MartyG-RealSense, 2020-10-12:
Hi @Andyshen555  As a starting point, you could test whether the camera is being permitted to observe depth values as far as possible by setting the launch flag **clip_distance** to a value of 10 (representing the default mximum depth sensing range of the 400 Series cameras).  This flag removes from the depth image all values above a given value in meters.

Andyshen555, 2020-10-12:
Thank you. I believe problem has been solved.

closed 2020-10-12 by Andyshen555

MartyG-RealSense, 2020-10-12:
Great to hear @Andyshen555 - thanks very much for the update!