realsenseai/realsense-ros #1445: RealSense T265 no topics published after ~1 minute without an error

Chris-Bee, 2020-10-14:
Hello everybody,

**System description**
Raspberry PI 4 64bit
Ubuntu 18.04
Kernel 5.4.65-v8
ROS Melodic
RealSense T265
RealSense ROS v2.2.17
Built with LibRealSense v2.39.0
Running with LibRealSense v2.39.0

To prevent power issues, the RealSense is connected to an external USB3 Hub with a standalone power supply [Digitus USB3 Hub](https://www.digitus.info/de/produkte/computer-und-office-zubehoer/computer-zubehoer/usb-komponenten-und-zubehoer/hubs-und-kartenleser/da-70231/)

**Error description:**
All topics published by the realsense-ros node stop publishing after a certain time usually 1-2 minutes. The realsense-ros node does not give me an error, journalctl and dmesg also do not show an error.

The interesting part is that, if this error happened and I restart the ROS node, then everything is publishing except for the two fisheye image streams. If I disconnect the RealSense USB connection, reconnect it, and start the ROS node again, then everything is publishing again. It is a bit difficult to debug this since there is no error message.

The only thing I noticed is that, if I change the logger level to info, I see the output below, and just before the topics stop being published, the output holds for a moment but then continues again. Thus, the debug messages still print "Fisheye stream published".

Let me know if there is another way on how I can further debug this issue.

**CMS output:**
~~~cmd
[DEBUG] [1602685147.929598927]: publishFrame(...)
[DEBUG] [1602685147.933108630]: Fisheye stream published
[DEBUG] [1602685147.941461464]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.943027667]: Publish Gyro stream
[DEBUG] [1602685147.944676408]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.946373945]: Publish Pose stream
[DEBUG] [1602685147.947880760]: Single video frame arrived (Fisheye, 2). frame_number: 1381 ; frame_TS: 1602685147196.118164 ; ros_TS(NSe
c): 1602685147210301399
[DEBUG] [1602685147.948502167]: publishFrame(...)
[DEBUG] [1602685147.951137019]: Fisheye stream published
[DEBUG] [1602685147.956071075]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.957553445]: Publish Gyro stream
[DEBUG] [1602685147.959836945]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.966166982]: Publish Pose stream
[DEBUG] [1602685147.967319112]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.968247223]: Publish Gyro stream
[DEBUG] [1602685147.970318741]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.971392204]: Publish Pose stream
[DEBUG] [1602685147.972393760]: Frame arrived: stream: Accel ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.973362630]: Publish Accel stream
[DEBUG] [1602685147.974421427]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.975093797]: Publish Gyro stream
[DEBUG] [1602685147.976285945]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.981208575]: Publish Pose stream
[DEBUG] [1602685147.982196890]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.982781093]: Publish Gyro stream
[DEBUG] [1602685147.983615371]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.985550964]: Publish Pose stream
[DEBUG] [1602685147.987054871]: Frame arrived: stream: Gyro ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.988542241]: Publish Gyro stream
[DEBUG] [1602685147.989671260]: Frame arrived: stream: Pose ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.990882186]: Publish Pose stream
[DEBUG] [1602685147.993139964]: Frame arrived: stream: Accel ; index: 0 ; Timestamp Domain: Global Time
[DEBUG] [1602685147.993943130]: Publish Accel stream
[DEBUG] [1602685147.995678593]: Single video frame arrived (Fisheye, 1). frame_number: 1382 ; frame_TS: 1602685147229.451660 ; ros_TS(NSe
c): 1602685147243634701
[DEBUG] [1602685147.997194723]: publishFrame(...)
[DEBUG] [1602685147.998985686]: Fisheye stream published
[DEBUG] [1602685148.007001871]: Single video frame arrived (Fisheye, 2). frame_number: 1382 ; frame_TS: 1602685147229.451660 ; ros_TS(NSe
c): 1602685147243634701
[DEBUG] [1602685148.007476315]: publishFrame(...)
[DEBUG] [1602685148.009598038]: Fisheye stream published
[DEBUG] [1602685148.017222278]: Single video frame arrived (Fisheye, 1). frame_number: 1383 ; frame_TS: 1602685147262.784912 ; ros_TS(NSe
c): 1602685147276968002
[DEBUG] [1602685148.017909593]: publishFrame(...)
[DEBUG] [1602685148.019637649]: Fisheye stream published
[DEBUG] [1602685148.023667001]: Single video frame arrived (Fisheye, 2). frame_number: 1383 ; frame_TS: 1602685147262.784912 ; ros_TS(NSe
c): 1602685147276968002
[DEBUG] [1602685148.024973204]: publishFrame(...)
[DEBUG] [1602685148.035498241]: Fisheye stream published
[DEBUG] [1602685148.038388390]: Single video frame arrived (Fisheye, 1). frame_number: 1384 ; frame_TS: 1602685147296.118408 ; ros_TS(NSe
c): 1602685147310301542
[DEBUG] [1602685148.038867538]: publishFrame(...)
[DEBUG] [1602685148.043551223]: Fisheye stream published
[DEBUG] [1602685148.045909353]: Single video frame arrived (Fisheye, 2). frame_number: 1384 ; frame_TS: 1602685147296.118408 ; ros_TS(NSe
c): 1602685147310301542
[DEBUG] [1602685148.047466982]: publishFrame(...)
[DEBUG] [1602685148.057936723]: Fisheye stream published
[DEBUG] [1602685148.388088371]: SteadyTimer deregistering callbacks.
[DEBUG] [1602685149.388070037]: SteadyTimer deregistering callbacks.
[DEBUG] [1602685150.390106203]: SteadyTimer deregistering callbacks.
^C[realsense/realsense2_camera-2] killing on exit
[realsense/realsense2_camera_manager-1] killing on exit

~~~

Chris-Bee, 2020-10-16:
Is this may be more of a question for the https://github.com/IntelRealSense/librealsense repository?

Chris-Bee, 2020-10-20:
I posted the issue on https://github.com/IntelRealSense/librealsense/issues/7620
I think that this issue is related to the driver due to the fact that the issue occurs when the throughput on the USB is high and requires a disconnect of the device to reset this problem. I guess this issue can be closed

Iko-Git, 2020-11-02:
I did face the same issue and did extensive debugging to find the problem. As based on the debugging messages, I ended up that the bellow _**default**_ part in  [_realsense-ros/realsense2_camera/src/base_realsense_node.cpp_](https://github.com/IntelRealSense/realsense-ros/blob/development/realsense2_camera/src/base_realsense_node.cpp) ​isn't joined by the _**switch (stream)**_ which belongs to 
https://github.com/IntelRealSense/realsense-ros/blob/b1c02bdf689c2e1b47c88fac098608c1e7964b91/realsense2_camera/src/base_realsense_node.cpp#L1687-L1703
As it's also shown in the same part of the code, we see how _stream_ is defined as following 
**_auto stream = frame.get_profile().stream_type();_**
Questioning what is **_frame_**?!
Particularly it's an object of **_frame_** class which belongs to **_rs2_** library (already in the link below) 
**_rs2::frame frame_**
http://docs.ros.org/en/kinetic/api/librealsense2/html/rs__frame_8hpp_source.html
Consequently, I see the issue now from API level

closed 2026-10-23 by MartyG-RealSense