realsenseai/realsense-ros #1446: No realsense2_node package to run

kylelscott, 2020-10-15:
Hello. I have a question about the project. After I build the repo, I do not have any packages like the one in the steps labeled "realsense_node". This leads to the error shown below. Do you know why this may be the case. I've followed the instructions to the tee. Thank you.

`kylel@kylel-desktop:~/ros2_ews$ colcon build
Starting >>> realsense2_camera_msgs
Finished <<< realsense2_camera_msgs [3.38s]                       
Starting >>> realsense2_camera
Finished <<< realsense2_camera [1.24s]                       

Summary: 2 packages finished [5.51s]
kylel@kylel-desktop:~/ros2_ews$ ros2 run realsense2_node realsense2_node
Package 'realsense2_node' not found
`

MartyG-RealSense, 2020-10-15:
Hi @kylelscott   I note the mention of ROS2 in your log.  If you are planning on using the official ROS2 wrapper for ROS2 Eloquent, did you follow the ROS2 version of the wrapper instructions in the link below, please?  

https://github.com/IntelRealSense/realsense-ros/tree/eloquent

MartyG-RealSense, 2020-10-22:
Hi @kylelscott  Do you still require assistance with this case, please?   Thanks!

kylelscott, 2020-10-22:
I apologize for the late reply. But yes I did follow these instructions.

MartyG-RealSense, 2020-10-22:
I could not find any past occurrence of this particular **Package 'realsense2_node' not found** error.   So I went through the instructions step by step.   It looks as though your problems begin after you have performed Step 5, **colcon build**, and then try to run realsense2_node with the command **ros2 run realsense2_node realsense2_node** once the build process has completed.

I note though that after **colcon build**, there is a Step 6 that does not seem to be inputted in your log of the terminal in the opening comment of this discussion.  This step comes directly before running realsense2_node.

**. install/local_setup.bash**

https://github.com/IntelRealSense/realsense-ros/tree/eloquent#step-6-source-on-each-new-terminal

kylelscott, 2020-10-22:
Ah, thank you. I did forget to source this. I am not available right now to verify, but I will when I am free. Thank you for your help.

MartyG-RealSense, 2020-10-22:
Thanks, I look forward to your update   :)

MartyG-RealSense, 2020-10-30:
Hi @kylelscott   Do you have an update about this case that you can provide please?   Thanks!

closed 2020-11-07 by MartyG-RealSense