realsenseai/realsense-ros #1447: need a settings file to connect T265 with ORB_SLAM3 on ROS platform

longzhenren, 2020-10-15:
I want to use realsense-ros to link the t265 camera and use ORB_SLAM3 for mapping. T265 can start successfully and the topic can be found in ROS, but the ORB_SLAM3 cannot be started because the dictionary and camera calibration file are need as parameters to be specified（ Asus.yaml in directory ORB_SLAM3/Examples/ROS/... ）Where can I get this settings file? Or what's wrong with my startup?
Ps:I used the command like "rosrun ORB_SLAM3 Stereo_Inertial PATH_TO_VOCABULARY PATH_TO_SETTINGS_FILE ONLINE_RECTIFICATION [EQUALIZATION]"
So how to get the settings file that PATH_TO_SETTINGS_FILE point to?

RealSenseSupport, 2020-11-08:
Hi,

If you're looking for the calibration parameter values to populate in the yaml file you can run "rs-enumerate-device.exe -c" to get that information. 

For specific help with ORB_SLAM we suggest opening a query on the ORB_SLAM3 repo as that is not an Intel product. Eg: this issue there seems like it might have useful info:

https://github.com/UZ-SLAMLab/ORB_SLAM3/issues/101

Thanks

longzhenren, 2020-11-09:
Thanks a lot！

closed 2020-11-09 by longzhenren