realsenseai/realsense-ros #1448: Log/warn when running in "reduced functionality" mode on USB2.0

breznak, 2020-10-15:
For D435i, 
it should run on USB3.0 for full functionality, if you plug into a USB2.0 port, the cam runs, but in "reduced mode". 

It would be nice if this is documented in more detail as what functionality is reduced? Also please log a warning, that "running in USB2 in reduced mode, and users should use USB3". 

Otherwise our build started to fail at runtime (ie 848x480@30Hz native resolution) is not supported. But cam did not print any err, only a warning about unsupported mode. The problem happened that someone re-plugged the cam into a usb2.

MartyG-RealSense, 2020-10-15:
Hi @breznak  Can you confirm please whether you are using the librealsense SDK or the RealSense ROS wrapper?   Thanks!

breznak, 2020-10-15:
Hi Marty, 

thank you for a quick response! 
We use the `ROS wrapper` (which in turn links to librealsense?). 

Log from the cam startup is: 
```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch.user]: Launching all available cameras
[INFO] [launch.user]: Configuration file used for cameras: rs_native.yaml
[INFO] [launch.user]: Found devices: [<pyrealsense2.device: Intel RealSense D435 (S/N: 936322070444)>]
[INFO] [launch.user]: Launching device with serial number 936322070444 in namespace /camera1.
[INFO] [realsense2_camera_node-1]: process started with pid [18151]
<REDACTED>
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: RealSense ROS v3.1.1
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Built with LibRealSense v2.38.1
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Running with LibRealSense v2.38.1
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Device with serial number 936322070444 was found.
[realsense2_camera_node-1]  [realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Device with physical ID /sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/0000:03:08.0/0000:07:00.3/usb4/4-4/4-4:1.0/video4linux/video0 was found.
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Device with name Intel RealSense D435 was found.
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Device with port number 4-4 was found.
[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Resetting device...
```

MartyG-RealSense, 2020-10-15:
Thank you very much for the confirmation that you are using the ROS wrapper.

As a starting point in trying to enable the camera at USB3 capability, could you add the term below to the end of your roslaunch instruction please:

**initial_reset:=true**

This causes a hardware reset of the camera to be performed at launch and sometimes resolves problems immediately.  So it is a good method to attempt first.

breznak, 2020-10-15:
> As a starting point in trying to enable the camera at USB3 capability 

just to clarify my problem. Everything works OK once I connected to USB3, my issue is with "silently degrading functionality" once conncted to usb2 - we should warn more explicitely about this. 

> initial_reset:=true
This causes a hardware reset of the camera to be performed at launch and sometimes resolves problems immediately. So it is a good method to attempt first

I believe we do that:
`[realsense2_camera_node-1] [INFO] [RealSenseCameraNode]: Resetting device...`
but I'll double check

MartyG-RealSense, 2020-10-15:
Thank you very much for the additional information to clarify your question.  I understand your meaning now: that you are knowingly establishing a USB2 connection but believe that there should be clearer documentation warning about the reduced camera functionality in this mode and perhaps a notification when inserting the camera that this will be the case.

There is already an information page about the limitations to expect when using USB2.  It can admittedly be difficult to find if you do not already know that it exists though.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

Regarding the subject of whether an in-wrapper warning about reduced functionality on USB2 could be displayed, I will refer that question to @doronhi the RealSense ROS wrapper developer.

breznak, 2020-10-15:
> that you are knowingly establishing a USB2 connection but believe that there should be clearer documentation warning about the reduced camera functionality 

maybe even unknowingly. In our case, someone needed to plug out the camera, and then plugged it back in, but to a usb2 port, not usb3 as before. This resulted in a silent break of our pipeline which was quite hard to pinpoint to the usb2 connection. 

So I suggest to be more explicit for the cause, if some functionality is unsupported due to running over usb2. 

> https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/ 

you're right, I didn't know about the page, thank you for the useful info! 

Thank you very much for providing help with the issue. 
Cheers,

MartyG-RealSense, 2020-10-23:
Hi @breznak  Do you still require assistance with this case, please?   Thanks!

breznak, 2020-10-23:
Thank you for checking up on this issue. I figured the problem myself, this issue is a placeholder to make the "degraded mode" more explicit, as I think it can be problem for others too. 
Best wishes,

MartyG-RealSense, 2020-10-23:
Okay, thanks very much for the update!   I will close the case as it will still be accessible via web search for future reference by others.   Thanks again!

closed 2020-10-23 by MartyG-RealSense