realsenseai/realsense-ros #1449: Transformation Matrices, D435i

iamsavva, 2020-10-15:
Hi. 
I have a few questions about reference frames.

1. I am looking for the transformation matrix between depth and infra camera poses. There must be a rotation + translation between these. How do I find it? Please help.

I checked the /camera/extrinsics/depth_to_infra_1 topic but it gives me an identity matrix. That's seems wrong from a physical point, am I missing something?

2. I am also looking for the transformations between IMU and infra cameras. According to [this website](https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/), IMU is published in the TF of the depth camera, quote: "IMU sample is multiplied internally by the extrinsic matrix". Is this correct? Is true that IMU measurements are in the depth frame? 

3. Infra images are rectified. Are they reprojected on the depth plane?

many thanks in advance for your help.

MartyG-RealSense, 2020-10-16:
Hi @iamsavva   IR is perfectly aligned to depth on the 400 Series cameras.  Intel's camera tuning guide for the 400 Series makes the following statement:

"The LEFT IR camera has the benefit of always 1. Being pixel-perfect aligned, calibrated, and overlapped with the depth map, 2. Perfectly time-synchronized, 3. Requires no additional computational overhead to align color-to-depth, and 4. Gives no additional occlusion artifacts, because it is exactly co-aligned". 

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#section-use-the-left-color-camera

2.  Yes, the IMU aligns  to depth.  The section in Intel's guide to IMU data that you linked to that is headed **Tracking sensor origin and coordinate system (CS)** describes the relationship between the IMU and depth:

"The IMU sensor location and orientation relative to the depth sensors are conveniently embedded into the sensor’s extrinsic data. In order to align and them with SDK-wide established orientation convention using the depth sensor as CS origin, each IMU sample is multiplied internally by the extrinsic matrix.  The resulted orientation angles and acceleration vectors share the coordinate system with the depth sensor.  The positive x-axis points to the right. The positive y-axis points down.  The positive z-axis points forward

[image]

3.  If you have not seen it already, I recommend reading the **Projection** documentation page to see whether it provides an answer to your question.

https://github.com/IntelRealSense/librealsense/wiki/Projection-in-RealSense-SDK-2.0

iamsavva, 2020-10-17:
thanks, that basically resolved it. Except for using external software packages like kalibr, are there any ways of determining the transformation matrices from imu to infra1, from imu to infra2? also, is there a CAD model of the camera that contains the location of the IMU? I would like to find the transformation matrix between the realsense camera and the robot that it is sitting on.

MartyG-RealSense, 2020-10-18:
I will refer your question about **transformation matrices from imu to infra1, from imu to infra2** to @doronhi the RealSense ROS developer.

The official RealSense CAD file archive does not contain a model for the D435i, just the D435.

https://dev.intelrealsense.com/docs/cad-files

The link below has an unofficial D435i CAD model in **SLDPRT** (SolidWorks Part File) format, though I do not know if the model contains that IMU location information that you require.

https://grabcad.com/library/intel-real-sense-d435i-camera-1

iamsavva, 2020-10-18:
> I will refer your question about **transformation matrices from imu to infra1, from imu to infra2** to @doronhi the RealSense ROS developer.
>
> The official RealSense CAD file archive does not contain [...]

thank you, Marty, I appreciate your help!
As far as the transformation matrices go, I've been able to acquire them with Kalibr. 
@doronhi , I guess I would be particularly interested in the specific location of the IMU on the D435i question more. It would be slightly challenging to rotate the robot around its axis.

doronhi, 2020-10-19:
The IMU's coordinate system, as reflected to the user, has the same orientation as the depth sensor but with some translation.
Using the `rs-enumerate-devices -c` command I get the following results:

```
...
Extrinsic from "Gyro"	  To	  "Depth" :
 Rotation Matrix:
   1   0   0
   0   1   0
   0   0   1

 Translation Vector: 0.00552000012248755  -0.00510000018402934  -0.011739999987185  
...
```

The same results can be obtained from the tf data while running realsense2_camera, using the following command: `rosrun tf tf_echo camera_depth_optical_frame camera_gyro_optical_frame`:

```
rosrun tf tf_echo camera_depth_optical_frame camera_gyro_optical_frame
At time 0.000
- Translation: [0.006, -0.005, -0.012]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
...
```

iamsavva, 2020-10-19:
@doronhi thank you for the reply and the clarification! Much appreciated, that's exactly what I needed.
One last question I had was in regard to the location of IMU in the camera or if you have any thoughts on finding transformations for the camera within the robot - if there is something you could add to Marty's reply. 

Cheers!

doronhi, 2020-10-20:
I assume you refer to the location of the IMU inside the camera's aluminum container.
The device is unaware of it's container therefor the information regarding it's mounting point is not available using the code.
The package realsense2_description contains the file "\urdf\_d435i.urdf.xacro" from which you can find the transformation between the bottom screw point and the imu.

If you run in one terminal: `roslaunch realsense2_description view_d435i_model.launch`
Then in another terminal you could `rosrun tf tf_echo camera_bottom_screw_frame camera_gyro_frame`.

The result is:
```
- Translation: [-0.001, 0.012, 0.018]
- Rotation: in Quaternion [0.000, 0.000, 0.000, 1.000]
            in RPY (radian) [0.000, -0.000, 0.000]
            in RPY (degree) [0.000, -0.000, 0.000]
```

iamsavva, 2020-10-27:
@doronhi, thank you for your help, this is perfect, precisely what I needed! Didn't realize this existed. Much appreciated! (apology for late reply).

closed 2020-10-27 by iamsavva

MartyG-RealSense, 2020-10-27:
I'm pleased that the advice of @doronhi was so helpful to you.   Thanks for the update!

Amiraqaie, 2023-02-01:
Hi @doronhi 
is there any way or formula to create the 4*4 matrix that represent the transformation between imu and infra1 fron the tf published on ros2? 
i have access to the translation and rotation of my camera to my imu but i dont know how to create the 4*4 body_T_cam matrix that needed by Vins-Fusion and i dont want to use Kalib package for some reasons.

MartyG-RealSense, 2023-02-01:
Hi @Amiraqaie  The only RealSense reference regarding **body_T_cam** other than this case is on the VINS-Fusion support forum at https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/6#issuecomment-706280995