realsenseai/realsense-ros #1451: How to sync rostopic time stamps among multiple devices?

haoyuanz13, 2020-10-16:
Here I have two d435 cams and launch them one by one, is there any approach to launch two d435 cams and their topics, e.g. depth image, has the exact same time stamp?

MartyG-RealSense, 2020-10-16:
Hi @haoyuanz13   It sounds as though you need to **hardware sync** the two cameras, so that one of the cameras (a "slave") follows the timing of another camera (the "master").  In the RealSense ROS wrapper, hardware sync can be accessed with the **inter_cam_sync_mode** option to set one camera as Master ('1') and the other camera as Slave ('2').

The subject was discussed on the RealSense ROS GitHub this week:

https://github.com/IntelRealSense/realsense-ros/issues/984#issuecomment-707709425

You can read much more about hardware sync between cameras in Intel's multiple camera sync white-paper document:

https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration

haoyuanz13, 2020-10-16:
Good to know that,  but still cannot figure out how to use this option via launch file ...
I have tried to add this arg into the .launch file and launch it, no error out but still get different time stamps between two cams topic ...

is there any more demos running like,
```bash
roslaunch realsense2_camera rs_xxxxx.launch xxxx

```

MartyG-RealSense, 2020-10-16:
Using hardware sync between cameras relies on creating a **sync cable** and joining the two cameras together.   There are a set of hardware sync pins concealed under a cover with a small hole on the top of the camera's casing.  The electronics diagram for creating a sync cable is in the multiple camera wite-paper document.

Andyshen555, 2020-10-18:
I believe the answer he want is: is there any way to set that parameter through launch file? That's also my question. Launching the dynamic config to tune the parameter of every camera every time after launching the camera is a stupid thing to do :(

MartyG-RealSense, 2020-10-18:
@Andyshen555  The subject of a launch file rosparam equivalent for inter_cam_sync_mode was discussed in a link from one of the pages that was linked to in this discussion.  So 2 links away.   :)

Please try the rosparam below and change '1' to the mode number that you want to set (0 for non-synced default, 1 for ,master, 2 for slave).

`<rosparam> /camera/stereo_module/inter_cam_sync_mode : 1 </rosparam>`

MartyG-RealSense, 2020-10-27:
Hi @haoyuanz13   Do you require further assistance with this case, please?   Thanks!

haoyuanz13, 2020-10-27:
Yes for sure, I have tried to add rosparam in the launch file, still not work correctly.
Now I am planning to try hardware sync via the cable.

Btw, could u pls provide one specific launch file with the `inter_cam_sync_mode` param?

MartyG-RealSense, 2020-10-27:
I researched your question carefully but could not find a pre-made launch file that features the **inter_cam_sync_mode** flag.

haoyuanz13, 2020-10-28:
Alright... so how can we make it work through this magic param?

MartyG-RealSense, 2020-10-28:
Use of a **rosparam** in a launch file to define the inter_cam_sync_mode setting is referenced in a comment earlier in this discussion:

https://github.com/IntelRealSense/realsense-ros/issues/1451#issuecomment-711430121

I have not seen inter_cam_sync_mode be set in the roslaunch launch instruction before (instead of putting it inside the roslaunch file as a rosparam).  If it were possible, I would expect the flag to add to the end of the roslaunch instruction to be something like this:

inter_cam_sync_mode:=2

again, change the number to the mode that you want to set (default, master, slave).

haoyuanz13, 2020-10-29:
alright, that makes sense somehow ...
Thx for your instruction!

MartyG-RealSense, 2020-11-05:
Hi @haoyuanz13   Do you still rquire assistance with this case, please?   Thanks!

haoyuanz13, 2020-11-06:
Maybe sometime in the future, but I will close this issue now.

closed 2020-11-06 by haoyuanz13