realsenseai/realsense-ros #1452: D435i align_depth colorizer

sunghoya1, 2020-10-16:
i used rs_aligned_depth.launch 
roslaunch realsense2_camera rs_aligned_depth.launch filters:=colorizer

the default depth image works fine, but the aligned depth image does not work in rqt with the following error:
[image]

I think this is probably conversion problem from U16C1->U8C3 
How to solve it? 
regar

MartyG-RealSense, 2020-10-16:
Hi @sunghoya1   May I confirm the steps that you are using before you encounter the errors.   Is it like this:

**roslaunch realsense2_camera rs_aligned_depth.launch filters:=colorizer
rosrun rqt_reconfigure rqt_reconfigure**

If this is the method that you are using, do you experience the errors if you do not use **filters:=colorizer** in the roslaunch instruction?

sunghoya1, 2020-10-16:
@MartyG-RealSense 
If filters:=colorizer is not used, it works fine with a black and white.

MartyG-RealSense, 2020-10-17:
Let's try a different approach to launch to get an aligned and colorized image, please.

roslaunch realsense2_camera rs_camera.launch align_depth:=true filters:=colorizer

doronhi, 2020-10-18:
This is a bug. Colorizer filter was not applied to aligned images.
Fixed in #1455

MartyG-RealSense, 2020-10-18:
Thanks very much for the information @doronhi   👍

sunghoya1, 2020-10-18:
@doronhi  @MartyG-RealSense 
It works perfectly. Thank you for your hard work

closed 2020-10-18 by sunghoya1