realsenseai/realsense-ros #1453: Gazebo plugin

sayyidabeegam, 2020-10-16:
Hello
I want to use D435 with my robot manipulator.
But I don't get any idea to integrate the sensor with my manipulator.
My requirements are
**1. I want to attach the D435 with my manipulator's base link
2. want to launch with my manipulator package
3. want to use all available sensor data format.**
Please help me to figure it out..
Thanks in advance

MartyG-RealSense, 2020-10-16:
Hi @sayyidabeegam   There is a Gazebo plugin for D435 at the link below.

https://github.com/pal-robotics/realsense_gazebo_plugin

Instructions for setting it up this plugin are here:

https://github.com/pal-robotics/realsense_gazebo_plugin/issues/7

A wider discussion about RealSense plugins for Gazebo in general is here:

https://github.com/IntelRealSense/realsense-ros/issues/1170

I hope that these resources will be a useful starting point for you.  Please do let me know of any follow-up questions that you have after reviewing the information.   Thanks!

sayyidabeegam, 2020-10-16:
Is it applicable only for ROS2?
I am using ROS 1 and by following this instruction I am getting error 
RLException:
Param xml is <param command="$(find xacro)/xacro $(arg model)" name="$(arg robot_name)/robot_description"/>
The traceback for the exception was written to the log file

MartyG-RealSense, 2020-10-16:
realsense_gazebo_plugin is based on the earlier SyrianSpock Gazebo plugin that worked with ROS1 Melodic, so I would think that the updated one should be Melodic compatible too.

https://github.com/SyrianSpock/realsense_gazebo_plugin

MartyG-RealSense, 2020-10-23:
Hi @sayyidabeegam   Do you still require assistance with this case, please?   Thanks!

sayyidabeegam, 2020-10-23:
Yes.
Still it is not working.
I am attaching the error message photo here with.
Looking forward to hearing from you
Thank you
Sayyida

sayyidabeegam, 2020-10-23:
yes
it is still not working

sayyidabeegam, 2020-10-23:
XML parsing error: mismatched tag: line 465, column 2
when processing file: /home/sayyida/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf/rx150.urdf.xacro

Check that:
 - Your XML is well-formed
 - You have the xacro xmlns declaration: xmlns:xacro="http://www.ros.org/wiki/xacro"
RLException: Invalid <param> tag: Cannot load command parameter [rx150/robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/sayyida/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf/rx150.urdf.xacro', 'robot_name:=rx150', 'use_world_frame:=true', 'use_default_gripper_bar:=true', 'use_default_gripper_fingers:=true', 'external_urdf_loc:=', 'load_gazebo_configs:=false']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro $(arg model)" name="$(arg robot_name)/robot_description"/>
The traceback for the exception was written to the log file

MartyG-RealSense, 2020-10-23:
Have you downloaded the two files in the links below and put those two files in the package inside the urdf folder where you want your RealSense camera to be simulated?

I wanted to make sure that you had managed to download them as the formatting of the file download links in the linked-to instructions were badly broken in my own web browser.

 https://github.com/pal-robotics-forks/realsense/blob/upstream/realsense2_description/urdf/_d435.gazebo.xacro

 https://github.com/pal-robotics-forks/realsense/blob/upstream/realsense2_description/urdf/_d435.urdf.xacro

Then follow the rest of the instructions, which I have edited slightly for ease of reading.

**Step 1**
Edit the file _**d435.urdf.xacro** and change line 13 from:

<xacro:include filename="$(find realsense2_description)/urdf/urdf_d435.gazebo.xacro"/>
to

<xacro:include filename="$(find packagename)/urdf/urdf_d435.gazebo.xacro"/>

**Step 2**
In the urdf file where you want to use the simulated realsense add the following code, while again replacing packagename with the name of the package you are using. and baselink which the link where you want the realsense camera to be joint on.

<xacro:include filename="$(find packagename)/urdf/_d435.urdf.xacro" />
<sensor_d435 parent="${prefix}base_link">
</sensor_d435>

**Step 3**
When you launch the file which uses this urdf model, you will find the simulated RealSense and you should see rostopics like **/camera/color/image_raw**

sayyidabeegam, 2020-10-24:
**Still error is there**

roslaunch interbotix_descriptions description.launch robot_name:=rx150 jnt_pub_gui:=true
... logging to /home/sayyida/.ros/log/179fbebe-15c9-11eb-a4c9-fef9b612eb5e/roslaunch-sayyida-11428.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Deprecated: xacro tag 'sensor_d435' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/sayyida/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf/rx150.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

Not enough blocks
when instantiating macro: sensor_d435 (/home/sayyida/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf/_d435.urdf.xacro)
in file: /home/sayyida/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf/rx150.urdf.xacro
RLException: Invalid <param> tag: Cannot load command parameter [rx150/robot_description]: command [['/opt/ros/melodic/lib/xacro/xacro', '/home/sayyida/interbotix_ws/src/interbotix_ros_arms/interbotix_descriptions/urdf/rx150.urdf.xacro', 'robot_name:=rx150', 'use_world_frame:=true', 'use_default_gripper_bar:=true', 'use_default_gripper_fingers:=true', 'external_urdf_loc:=', 'load_gazebo_configs:=false']] returned with code [2]. 

Param xml is <param command="$(find xacro)/xacro $(arg model)" name="$(arg robot_name)/robot_description"/>
The traceback for the exception was written to the log file
**NB:There is no problem with arm package Independently. The error come only after giving the realsense plugin**

MartyG-RealSense, 2020-10-24:
I do not have any personal experience with Gazebo or the RealSense gazebo plugin in particular, so I had to rely on references created by others.  The best references that I could find were these two links:

https://roboticsknowledgebase.com/wiki/tools/gazebo-simulation/
https://answers.ros.org/question/348331/realsense-d435-gazebo-plugin/

sayyidabeegam, 2020-11-02:
No

closed 2020-11-02 by sayyidabeegam

MartyG-RealSense, 2020-11-02:
Thanks very much for the update!

rickstaa, 2020-12-01:
For future reference [here](https://github.com/rickstaa/realsense-ros-gazebo) is an example package on how to use this plugin with gazebo and ROS, might people need it. A detailed explanation video created by [@issaiass](https://github.com/issaiass) can  be found [here](https://www.youtube.com/watch?v=hpUCG6K5muI&feature=youtu.be).

MartyG-RealSense, 2020-12-01:
Awesome, thanks so much @rickstaa for sharing the resources with the RealSense community!