realsenseai/realsense-ros #1454: IMU Timestamps are strange

iamsavva, 2020-10-17:
Hi,

I am using d435i camera on a TX2. d435i is runing on firmware 5.12.7.100. I installed `realsense2_camera` via `apt-get install`. I am running the following roslaunch statement:
`roslaunch realsense2_camera rs_camera.launch enable_infra1:=true enable_infra2:=true infra_fps:=15 enable_gyro:=true enable_accel:=true unite_imu_method:=linear_interpolation enable_sync:=true
`
i noticed that imu timestamps have a very wrong header. I attached a screenshot. From left to right you can see the outputs when I echo /camera/imu/header, /camera/infra1/image_rect_raw/header, /camera/infra2/image_rect_raw/header.

What is wrong with the timestamps? I tried every combination of launch file settings possible, the timestamps are always off 50 years ahead.

[image]

MartyG-RealSense, 2020-10-18:
Hi @iamsavva   I used a **convert seconds to years** tool and converted the timestamp value **1602971435** from your image above.  It provided a result of 50.8 years, so I can see how you arrived at your figure.

I ran thorough tests with a D435i and got similar timestamp results from the metadata.  I think that where the confusion may be occurring is that the timestamp is in **milliseconds** and not seconds.  So a timestamp of **1602971435** (milliseconds) converts to **0.05** years or 18 days.

[image]

In my thorough tests stopping the IMU stream and restarting it reset the frame number to zero but the timestamp continued incrementing from roughly where it left off when the previous streaming session was closed.

doronhi, 2020-10-18:
My guess is no metadata is available for the IMU frames. Therefor it transmits system time for the IMU where the other frames have hardware timestamps.

iamsavva, 2020-10-18:
> I think that where the confusion may be occurring is that the timestamp is in **milliseconds** and not seconds. So a timestamp of **1602971435** (milliseconds) converts to **0.05** years or 18 days.

Hi Marty, I think this is wrong and the 1602971435 is in fact in seconds - because if I plug into the the [epoch converter](https://www.epochconverter.com/), I get exactly the date and time that I was running the tests. And the 18 days number doesn't make much sense to me.

So I did find the source of the problem and the solution. If I plug the camera and `roslaunch` it the first time - the timestamps on IMU are correct. If I then `ctrl-c `the `roslaunch` and restart it again - that's when the timestamps go 50 years ahead (32... instead of 16...). Restarting the `roslaunch` again and again doesn't change the timestamps to (48...), they stay at (32...). If I unplug the camera and plug it back in, the timestamp gets set back to the correct one (16..) again. If I use the `initial_reset:=true` flag every time I `roslaunch`, the timestamps also get reset to the proper (16...).  So it must something related to the ros code inappropriately quitting execution. 

This solution works for me, but it feels like a hack, plus the initial_reset is pretty slow. Would appreciate a fix:)
Thank you for your help!

closed 2020-10-18 by iamsavva

MartyG-RealSense, 2020-10-18:
Timestamps are expressed in milliseconds when retrieved by **get_timestamp()**:

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1

18 days would have made sense in the context of a camera that does not reset the timestamp after the stream is closed and continues accumulating during multiple camera sessions over a long period of time.  The comment linked below though validates your point about a relationship to epoch:

https://github.com/IntelRealSense/librealsense/issues/4525#issuecomment-515951423

I'm pleased that you found a method that works for you!

JNimkar, 2020-12-04:
Hi, 

I am having this exact same issue. However, @iamsavva's solution of adding initial_reset:=true does not work for me. Would appreciate any help in resolving this issue. Thanks! 

[image]

MartyG-RealSense, 2020-12-05:
Thanks very much @JNimkar for sharing what worked for you (above) with the RealSense ROS community - great to hear that you found a solution!