realsenseai/realsense-ros #1456: main stream

atinfinity, 2020-10-18:


MartyG-RealSense, 2020-10-18:
Hi @atinfinity   The ROS1 wrapper is the most well-developed because it has had multiple releases over a number of years.  The ROS2 wrapper, which target ROS Eloquent, is relatively new at the time of writing this.   ROS2 is considered as important as ROS1, so whilst its wrapper is recent, it should not be thought of as being less mainstream than ROS1.

The discussion in the link below gives a good general overview of plans for ROS2 by the RealSense ROS wrapper developer @doronhi 

https://github.com/IntelRealSense/librealsense/issues/5825

atinfinity, 2020-10-18:
@MartyG-RealSense Thank you for your reply. I'll read this issue.

closed 2020-10-18 by atinfinity

JaehyunShim, 2020-10-27:
@MartyG-RealSense @atinfinity
It looks like the question is why there are two separate repositories for ROS2 packages (or which one is the mainstream). As far as what I can see from the contributor list, the second repository was developed by the intel team in China but it is not maintained at all.

JaehyunShim, 2020-10-27:
@MartyG-RealSense I mentioned it already in another issue thread, but if you guys are not going to maintain it, archive it and guide users to the mainstream repository (the first link) and send PRs there, not to the deprecated one.

MartyG-RealSense, 2020-10-27:
Hi @JaehyunShim   The currently developed ROS2 wrapper is the one in the link below, which supports **ROS2 Eloquent**:

https://github.com/IntelRealSense/realsense-ros/tree/eloquent

I believe that the wrapper with dormant development that you refer to is this one:

https://github.com/intel/ros2_intel_realsense

From the URL address, it looks as though it is hosted on the general **Intel** GitHub site, not the dedicated **RealSense ROS** GitHub.  RealSense projects on the main Intel GitHub do occasionally occur, an example of this being a project to display RealSense depth streams in a web-browser.

https://github.com/intel/depth-camera-web-demo

The RealSense ROS team would not have control over the content hosted on the main Intel GitHub.