realsenseai/realsense-ros #1458: Data loss when using rosbag record

Grandzxw, 2020-10-19:
I want to record the three topic: color/depth/imu, but when I use rosbag record, the data of image will lost, what’s the problem??

MartyG-RealSense, 2020-10-19:
Hi @Grandzxw   Do you mean that you are missing some frames compared to how many frames you think the bag should contain, please?

Grandzxw, 2020-10-19:
> Hi @Grandzxw Do you mean that you are missing some frames compared to how many frames you think the bag should contain, please?
> 
> Or do you have some streams missing completely from the recording, such as IMU data?

YES,  it lost a lot of frames

Grandzxw, 2020-10-19:
> YES, it lost a lot of frames

 It lost a lot of depth frames

MartyG-RealSense, 2020-10-19:
Thank you very much for the confirmation.  What frames per second (FPS) speed are you recording at, please?  And are you recording on a PC computer or a board such as a Jetson?

Grandzxw, 2020-10-20:
> Thank you very much for the confirmation. What frames per second (FPS) speed are you recording at, please? And are you recording on a PC computer or a board such as a Jetson?

I record on PC. And the depth frame is at 6Hz, the color frame is 10Hz. So it lost much data. They were at 30 Hz.

Grandzxw, 2020-10-20:
> Thank you very much for the confirmation. What frames per second (FPS) speed are you recording at, please? And are you recording on a PC computer or a board such as a Jetson?

Do you know how to solve it？

MartyG-RealSense, 2020-10-20:
As a starting point, I would recommend testing recording depth at an FPS speed of at least 10 (preferably 15).  Recording at 6 FPS can cause problems, such as time-outs because new frames are not arriving quickly enough.

closed 2020-11-07 by MartyG-RealSense