realsenseai/realsense-ros #1460: Clarification: Latest development branch compatibility (ros1? ros2? Ubuntu 16.04? 18.04?)

130s, 2020-10-19:
I couldn't find relevant announcement/discussion so asking here for clarification.

We noticed in [realsense-ros/releases](https://github.com/IntelRealSense/realsense-ros/releases) that the latest releases are version 3.x, which say supported Ubuntu version is 18.04. Specifically we have the following questions that we'll appreciate to be clarified:

- Is Ubuntu 16.04 NOT going to be getting bufixes and/or new features?
- Which branch is the development branch for 16.04? [development](https://github.com/IntelRealSense/realsense-ros/tree/development)?

CC @abhijitmajumdar

doronhi, 2020-10-20:
Thank you for your question.
The versions 3.x are meant for ROS2-eloquent and as such support Ubuntu 18.04.
The versions 2.x are for ROS (ROS1) and support Ubuntu 16.04 and 18.04.

As for your questions:
- The version for ROS (ubuntu 16.04) is still being developed and maintained. Bugs are being fixed and new features will be added as needed.
- the [development](https://github.com/IntelRealSense/realsense-ros/tree/development) branch is, as, you said, the development branch for ROS1 (therefor - Ubuntu 16.04). The [eloquent](https://github.com/IntelRealSense/realsense-ros/tree/eloquent) branch is the development branch for ROS2-eloquent.

130s, 2020-10-20:
Thanks for prompt clarification.

I'd usually request (or send a PR for) this sort of info to be documented but assuming the things are in temporary state I'm fine in just closing this ticket.

closed 2020-10-20 by 130s

doronhi, 2020-10-21:
Although this may be a temporary state, I would also like to have this documented.
If you have the time, I would like to hear suggestions or have a PR after all.
In the [development branch's readme.md file](https://github.com/IntelRealSense/realsense-ros/blob/development/README.md) it says that "For running in ROS2 environment please switch to the [eloquent branch](https://github.com/IntelRealSense/realsense-ros/tree/eloquent)".
Do you find that comment not clear enough?
Regarding the version enumeration (2.x for ROS1, 3.x for ROS2) where, in your opinion, will be a good place to write down an explanation?