realsenseai/realsense-ros #1462: Installation issue Ubuntu 18.04.5

ELEGER, 2020-10-21:
Hi,

Thanks for developping a sdk on linux plateform.

I'm trying to install the sdk on a ubuntu 18.04.5 LTS, following [this](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). 
I'm failing in step 4, "Building and applying the patch", 
`./scripts/patch-realsense-ubuntu-lts.sh` 
gives:

> Package required libusb-1.0-0-dev:  - found

> Package required libssl-dev:  - found
> Create patches workspace in ubuntu-bionic-5 folder

> Package required libelf-dev:  - found
> Package required elfutils:  - found
> Package required bison:  - found
> Package required flex:  - found
> Fetching Ubuntu LTS tag   to the local kernel sources folder
> fatal: You need to specify a tag name.

For info, Kernel:
`5.0.0-1070-oem-osp1 #76-Ubuntu SMP`

Thanks for your help

MartyG-RealSense, 2020-10-21:
Hi @ELEGER   It looks as though the problem is likely due to an unusual kernel version - **5.0.0-1070-oem-osp1** - being used.  Librealsense prefers 'LTS' kernels, though 'generic' ones work too.

If you cannot change your kernel then you can work around it.  There is a method of installing the SDK from source code called **RSUSB** that is not dependent on kernel versions or Linux versions and does not require patching (though the installation process does require an internet connection).

The RSUSB installation process is fully described in the link below:

https://github.com/IntelRealSense/librealsense/issues/6368#issuecomment-636316279

ELEGER, 2020-10-21:
Thanks for your quick answer. I'll try the RSUSB way.

Thanks

ELEGER, 2020-10-21:
It went through.

For the sake of completeness:

`cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true`

Thank you for your efficiency,

Best

closed 2020-10-21 by ELEGER

MartyG-RealSense, 2020-10-21:
Great to hear that it worked for you - thank you for the update!