realsenseai/realsense-ros #1463: Wrong ROS timestamp for published messages after turning on IMUs

dorianhenning, 2020-10-21:
### Wrong timespaming after enable IMUs

So if I launch the realsense node with the provided launch file `rs_camera.launch`, everything looks perfect and I get the following ROS message for `/camera/color/camera_info`:

```
header: 
  seq: 124
  stamp: 
    secs: 1603300835
    nsecs: 516419411
  frame_id: "camera_color_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [-0.05795837938785553, 0.06494612991809845, 0.0004481421201489866, 0.000659768411424011, -0.02090403623878956]
K: [383.4322509765625, 0.0, 314.5206298828125, 0.0, 383.08941650390625, 254.92738342285156, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [383.4322509765625, 0.0, 314.5206298828125, 0.0, 0.0, 383.08941650390625, 254.92738342285156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

However, if I change the two parameters `enable_gyro` and `enable_accel` to `true`,  I get the following message:

```
header: 
  seq: 247
  stamp: 
    secs: 3206600798
    nsecs: 350492001
  frame_id: "camera_color_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [-0.05795837938785553, 0.06494612991809845, 0.0004481421201489866, 0.000659768411424011, -0.02090403623878956]
K: [383.4322509765625, 0.0, 314.5206298828125, 0.0, 383.08941650390625, 254.92738342285156, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [383.4322509765625, 0.0, 314.5206298828125, 0.0, 0.0, 383.08941650390625, 254.92738342285156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

As one can see, the timestamps of the second message are clearly wrong:
 **stamp: 
    secs: 3206600798
    nsecs: 350492001**
compared to
  **stamp: 
    secs: 1603300835
    nsecs: 516419411**

Does anyone know a solution to that?

MartyG-RealSense, 2020-10-22:
Hi @dorianhenning   There was a timestamp related case in the past week that reminded me of your results, regarding jumpng from '16' to '32'.  A solution in that case was to include **initial_reset:=true** in the roslaunch statement to perform a hardware reset of the camer at launch.

https://github.com/IntelRealSense/realsense-ros/issues/1454#issuecomment-711172609

dorianhenning, 2020-10-22:
@MartyG-RealSense I tried that parameter, but unfortunately does this not solve the problem. It feels like a very random process and I cannot find a rule on how it works when I try different combinations of unplugging and using the parameter **initial_reset**, but in every case I either get _correct timestamps and no IMU measurements_, or _no measurements but the correct timestamps_.

MartyG-RealSense, 2020-10-22:
Further research found another case where problems with the timestamp doubling were being encountered when using the IMU, and the RealSense user in that case found that disabling **Global Time** and using the Hardware Clock seemed to be a workaround.

https://github.com/IntelRealSense/librealsense/issues/5067

shuhaowu, 2020-10-25:
I can confirm that (if rs2_camera is launched under the `/camera` namespace) running `rosrun dynamic_reconfigure dynparam set /camera/motion_module global_time_enabled false` will fix the timestamp of `/camera/imu`.

Is there a way to make this setting permanent? I tried to add a node in my roslaunch file, but it doesn't work consistently as it races with the startup of the realsense node, which sometimes will set the global_time_enabled back to true:

```
    <node name="realsense_fix_global_time"
          pkg="dynamic_reconfigure"
          type="dynparam"
          args="set /camera/motion_module global_time_enabled false"
          output="screen" />

```

MartyG-RealSense, 2020-10-25:
Each dynamic reconfigure parameter in the RealSense ROS wrapper has a <rosparam> equivalent with the same function name that can be placed in the roslaunch file.

The formatting for the rosparam equivalent of a dynamic reconfigure function is:

`<rosparam> /<camera topic>/<function name>: <state to set function to> </rosparam>`

So if **/camera/motion_module** is the topic, **global_time_enabled** is the function name and it is being set to the state of **false** then I believe the rosparam instruction in the launch file would look like this:

`<rosparam> /camera/motion_module/global_time_enabled: false </rosparam>`

MartyG-RealSense, 2020-11-02:
Hi @dorianhenning   Do you still require assistance with this case, please?  Thanks!

dorianhenning, 2020-11-06:
sorry, but I cannot reproduce/confirm @shuhaowu 's solution. I still get the wrong timestamps. Also, I believe this is something that should be the case for the out-of-the-box launch files, etc.

MartyG-RealSense, 2020-11-06:
@dorianhenning   What happens if you add onto the end of your roslaunch instruction the term below to unite the two individual gyro and accel topics into a single topic called **imu** that is published at the rate of the gyro?

**unite_imu_method:=linear_interpolation**

doronhi, 2020-11-08:
I can't reproduce. Could you please specify which firmware do you use, which camera model, firmware version, librealsense2 version and realsense2_camera version? Also the log of realsense2_camera - it might contain some details that I neglected to specify here.

MartyG-RealSense, 2020-11-15:
Hi @dorianhenning   Do you require further assistance with this case, please?   Thanks!

dorianhenning, 2020-11-15:
Hi Marty, I don’t have time to work on this at the moment, so I will come back in a month or so and try it again. Thank you anyways with your assistance!

closed 2020-11-15 by dorianhenning

MartyG-RealSense, 2020-11-15:
Thanks very much for the update @dorianhenning      :)