realsenseai/realsense-ros #1464: Unable to locate package

limitedboom, 2020-10-21:
I tried running the line of sudo apt-get install ros-$ROS_VER-realsense2-camera. But get the error of "Unable to lcoate package ros--realsense2-camera. would anyone know why this would be happening because I tried different githubs for intel realsense installation but get this same error one way or another when installing.

MartyG-RealSense, 2020-10-22:
Hi @limitedboom   Can you confirm please whether you are using the Raspbian or the Ubuntu OS on your Pi 4?  The official installation instructions are suited for use with Ubuntu.  

If you are using Raspbian then I would recommend a librealsense + ROS installation guide designed for Raspbian.

A good example of such a guide is the **eleccelerator** one, which explains errors that might occur at each step and fixes for them.

**eleccelerator**
https://eleccelerator.com/wiki/index.php?title=Raspbian_Buster_ROS_RealSense

limitedboom, 2020-10-22:
I am using Ubuntu.

MartyG-RealSense, 2020-10-23:
Thank you very much for the confirmation of the OS that you are using.

I believe that the best source of information for your situation (Pi 4 with Ubuntu trying to install ROS and the RealSense ROS wrapper) will be the instruction guide contributed by a RealSense user in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/360049881094/comments/360014049813

closed 2020-11-07 by MartyG-RealSense