realsenseai/realsense-ros #1466: D435i Sync problem

YangTiankai, 2020-10-26:
Set the FPS to 30Hz, and set the param "Inter Cam Sync Mode" to 33 in realsense-viewer. The actual camera frame rate can't reach 30Hz

Set the FPS to 30Hz, and set the param "Inter Cam Sync Mode" to 32 in realsense-viewer. The actual camera frame rate is 29Hz. And the interval of two frame is 33.33ms, but one frame is missing every second.

The PPS signal is generated by DJI N3 filght control module.

I hope the pps signal can trigger a 30hz camera frame rate.
I want to know if this is caused by the actual frequency of the PPS signal being slightly higher than 1Hz.

MartyG-RealSense, 2020-10-26:
Hi @YangTiankai   By setting  **Inter Cam Sync Mode** to 32 and 33 respectively, I believe that your frames are being affected by the **External Synchronization (Genlock)** function of the RealSense SDK.

https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-5-programming-camera-sync

The purpose of **Inter Cam Sync Mode** is to synchronize frame timestamps with a trigger pulse, not sync FPS speeds.  So if your goal is to force FPS to try to remain at 30 constantly instead of fluctuating then you likely do not need to use sync at all and the Inter Cam Sync Mode should be set to the default of '0' (no sync).

If you want to force FPS to remain at the value it was set at (e.g 30 FPS), you can do so either by one of the following methods:

1.  Disabling auto-exposure and setting a manual exposure value that is within a certain range, or 

(2)  By having auto-exposure enabled and **auto-exposure priority** disabled.  The auto-exposure priority control in the RealSense Viewer is located under the RGB controls in the Viewer's options side-panel.

Further details of these two methods for forcing a constant FPS are given in the link below.

https://github.com/IntelRealSense/librealsense/issues/1957#issuecomment-400715579

YangTiankai, 2020-10-26:
I see that
"This does mean that it is possible to approach the full frame rate, in this case 90fps, with a 1Hz trigger, by setting the native camera mode to 90 fps and setting the count to 89 while triggering at 1Hz, for example. "
in
[https://dev.intelrealsense.com/docs/external-synchronization-of-intel-realsense-depth-cameras#section-5-programming-camera-sync](url)

It seems that the third situation in the figure below cannot be achieved. 

Thanks for your advice. I think I can set the FPS to 60 and the sync mode to 62, and get rid of the 29 frames to reach a 30FPS image stream

[image]

MartyG-RealSense, 2020-10-26:
Thnaks very much for the update.  Please do post further comments if you still require assistance after your tests.  Good luck!

MartyG-RealSense, 2020-11-02:
Hi @YangTiankai   Do you have an update that you can provide me with please?  Thanks!

closed 2020-11-08 by MartyG-RealSense