realsenseai/realsense-ros #1467: Error while building on Eloquent

rafaelRinconFo, 2020-10-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.4.0-52                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.16.5 }                          | 
| Language                            |  Docker                         | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi guys!

Currently, I'm trying to install the Ros2 Eloquent package within a Docker File. When Building, I get the following log:

`--- stderr: realsense2_camera                                                                                                                                           
CMake Warning (dev) at /usr/local/share/cmake-3.15/Modules/FindCUDA.cmake:580 (option):
  Policy CMP0077 is not set: option() honors normal variables.  Run "cmake
  --help-policy CMP0077" for policy details.  Use the cmake_policy command to
  set the policy and suppress this warning.

  For compatibility with older versions of CMake, option is clearing the
  normal variable 'CUDA_PROPAGATE_HOST_FLAGS'.
Call Stack (most recent call first):
  /usr/local/lib/cmake/opencv4/OpenCVConfig.cmake:86 (find_package)
  /usr/local/lib/cmake/opencv4/OpenCVConfig.cmake:108 (find_host_package)
  /opt/ros/dashing/share/cv_bridge/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/dashing/share/cv_bridge/cmake/cv_bridgeConfig.cmake:38 (include)
  CMakeLists.txt:89 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning at CMakeLists.txt:102 (find_package):
  Could not find a configuration file for package "realsense2" that is
  compatible with requested version "2.38.1".

  The following configuration files were considered but not accepted:

    /opt/ros/dashing/lib/cmake/realsense2/realsense2Config.cmake, version: 2.16.5

CMake Error at CMakeLists.txt:104 (message):
  

  

   Intel RealSense SDK 2.0 is missing, please install it from https://github.com/IntelRealSense/librealsense/releases

  

`

I have already installed the SDK as stated in [https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md](url) . However, I keep getting that log. I don't know if the fact of it being a Docker Container affects the package building when executing the CMake. 

I look for your response.

MartyG-RealSense, 2020-10-27:
Hi @rafaelRinconFo   As a starting point in investigating your problem, I note that you list librealsense version **2.16.5** from November 2018 as the librealsense SDK version that you are using.  ROS wrapper versions should be matched as closely as possible with the librealsense version number listed on the Release Notes for a particular wrapper version.  

https://github.com/IntelRealSense/realsense-ros/releases

The first version of the official ROS2 Eloquent wrapper began with a recommendation of SDK **2.38.1**, and the latest ROS2 Eloquent wrapper at the time of writing this recommends librealsense **2.39.0**.

rafaelRinconFo, 2020-10-27:
Hi @MartyG-RealSense , is there any way to force the installation of a specific version of the SDK? I have already performed the ` sudo apt-get --only-upgrade install librealsense2-utils librealsense2-dkms` command and `sudo apt-get` commands as recommended and the resulting log states that librealsense2-dkms and librealsense2-utils are already in the latest versions.

MartyG-RealSense, 2020-10-27:
I hope that the link below will provide the guidance that you need for installing specific versions of packages.  It recommends removng the currently installed SDK packages and then guides about how to install particularly-numbered packages:

https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

doronhi, 2020-10-28:
I think the issue might be that librealsense2 2.38.1 is not yet available in the ros-distributions for eloquent. There is a merged PR for it but they have not released an update yet. 
In the meanwhile, I think you'll have to install it separately, following the [the librealsense2installation guide for linux](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) it may find version 2.38.1 there.

closed 2020-11-13 by MartyG-RealSense