realsenseai/realsense-ros #1469: D435i imu data lossing

zhizunhu, 2020-10-28:
I have two realsense d435i. When I open the dirver with imu interpolation on, I found that both of them have imu data lost
Hear is my config yaml:
[image]
And I record a rosbag:
[image]

MartyG-RealSense, 2020-10-28:
Hi @zhizunhu   Do you still experience IMU data loss if **unite_imu_method** is not used?

Also, could you confirm please the extent of the missing data.  Is it dropped frames, or no IMU data at all?

zhizunhu, 2020-10-28:
Hi @MartyG-RealSense Yes, I close unite_imu_method, It still has loss:
[image]
And the freq of gyro is not consistent like that (the mid is the sample of gyro)：
[image]
 And I found that there is no loss in visual frames.

MartyG-RealSense, 2020-10-28:
Thank you very much for the charts.  So the large white gaps in the gyro image represent missing frames that should be there?

zhizunhu, 2020-10-29:
Yes @MartyG-RealSense , each blue line means one sensor sample. They are placed by timestamp. Big white gaps means small freq. The imu datas of D435i I tested were not sampled by a consistent freq. 
Then I just play the rosbag recorded and echo the imu topic, found that there is some loss like that:
[image]
[image]
And it always lose 3 samples. But the seq was added, Does it mean that the driver has solved these data but not published?

MartyG-RealSense, 2020-10-29:
Does adding the flag **initial_reset:=true** to your roslaunch statement to perform a hardware reset of the camera at launch improve the results, please?

skiddattle, 2020-10-30:
I think I may have a similar issue, where the /camer/imu topic freezes every now and then. I am using ubuntu 18.04 on a laptop.

Here is a youtube link to the freezing output: https://youtu.be/PkUiWxzqiK4

I have tried initial_reset:=true and it doesn't improve performance.
I have also checked to accel and gyro topics to see if there is any freezing and it doesnt show any freezing.

For unite imu method I used linear_interpolation.

It will update, and freeze at a certain point, and then update again after a while, and this cycle repeats.

**Command:**
```
rostopic echo /camera/imu
```
**Console output:**
```
header: 
  seq: 25295
  stamp: 
    secs: 1604058751
    nsecs: 334259987
  frame_id: "camera_imu_optical_frame"
orientation: 
  x: 0.0
  y: 0.0
  z: 0.0
  w: 0.0
orientation_covariance: [-1.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity: 
  x: 8.73358385434e-06
  y: -0.00338855828159
  z: -0.00173951964825
angular_velocity_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
linear_acceleration: 
  x: -0.0595852163378
  y: -9.80048897499
  z: 0.289907492512
linear_acceleration_covariance: [0.01, 0.0, 0.0, 0.0, 0.01, 0.0, 0.0, 0.0, 0.01]
```

**Command:**
```
roslaunch realsense2_camera rs_camera.launch
```

**Console output:**
```

bryan@bryan-ThinkPad-T450:~$ roslaunch realsense2_camera rs_camera.launch
... logging to /home/bryan/.ros/log/cc1f7d4e-1aa5-11eb-893b-94659c228819/roslaunch-bryan-ThinkPad-T450-6104.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://bryan-ThinkPad-T450:33423/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
[... 223 lines cut ...]
      <arg name="depth_fps"                value="$(arg depth_fps)"/>
      <arg name="infra_fps"                value="$(arg infra_fps)"/>
      <arg name="color_fps"                value="$(arg color_fps)"/>
      <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
      <arg name="accel_fps"                value="$(arg accel_fps)"/>
      <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
      <arg name="enable_accel"             value="$(arg enable_accel)"/>

      <arg name="publish_tf"               value="$(arg publish_tf)"/>
      <arg name="tf_publish_rate"          value="$(arg tf_publish_rate)"/>

      <arg name="filters"                  value="$(arg filters)"/>
      <arg name="clip_distance"            value="$(arg clip_distance)"/>
      <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
      <arg name="initial_reset"            value="$(arg initial_reset)"/>
      <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
      <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
      <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
      <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
      <arg name="allow_no_texture_points"  value="$(arg allow_no_texture_points)"/>
    </include>
  </group>
</launch>

```

skiddattle, 2020-10-30:
To get it to stop freezing, I set the gyro and accel to lower fps and it seems to stop the freezing issue: 
  <arg name="gyro_fps"            default="200"/>
  <arg name="accel_fps"           default="63"/>

The curious thing is that I have the same setup in a different environment which is on the jetson xavier nx running ubuntu 18.04 but it does not have the freezing issue when I set it to the rates of 400hz and 250hz respectively. Is this an issue with the installation?

skiddattle, 2020-10-30:
I would like to see if I can resolve this issue while maintaining the gyro and accel rates at the maximum possible of 400 and 250, as I would like to use it for visual inertial odometry and it is important for accuracy.

MartyG-RealSense, 2020-11-07:
Hi @skiddattle  Do you require further assistance with this case, please?   Thanks!

skiddattle, 2020-11-08:
No need, thanks!

MartyG-RealSense, 2020-11-08:
Thanks very much @skiddattle 

Hi @zhizunhu    As you created this case, do you still need help?

ericpairet, 2020-11-15:
Hi @MartyG-RealSense,

I observe the same issues as reported by @zhizunhu on a Raspberry Pi 4. On top of that, sometimes the ROS wrapper stops publishing IMU data indefinitely despite the node remains alive (permanent blackout). As mentioned by @zhizunhu, there's no loss on the visual frames.

My setup: I initially intended to use [Intel's official pre-built image for the Pi](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras?-697455189.1592827118&_ga=2.21903602.813603083.1605425812-70032516.1601732816#section-2-3-preparing-the-sd-card) but I could not get the ROS wrapper working properly. Then, I tried two different setups of (librealsense - ROS wrapper): (2.38 - 2.2.17) and (2.39 - 2.2.18), following [these instructions for the librealsense installation on the Pi](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md). Both setups seem to run, but with IMU data loss and, eventually, permanent blackout. 

I tried reducing the FPS of all devices to the minimum, just leaving on the accel and gyro (without fusing the data, and fusing their data with `unite_imu_method=copy` and `unite_imu_method=linear_interpolation`), and to force an `initial_reset=true` at startup. Any of these tests resulted in a consistent flow of IMU data, and unfortunately, I couldn't see any pattern.

Thanks in advance for any help!

MartyG-RealSense, 2020-11-15:
Hi @ericpairet 

@skiddattle above had the same problem, and did not require further assistance after being given a link that explained about IMU frequencies.  Have you visited this link too please?

https://github.com/IntelRealSense/realsense-ros/issues/729

ericpairet, 2020-11-16:
Hi @MartyG-RealSense,

Thanks for your fast response. I checked #729, but it doesn't clarify to me the temporal and permanent blackouts on the gyro, accel and imu topics. As both issues also happen with the raw data from the gyro and accel, I believe it might not be a problem related to `unite_imu_method`. I cannot replicate the issues on my desktop machine (the ROS wrapper works flawlessly in there), so if you think that the problem might be related with the Raspberry Pi setup, I'm happy to open a new ticket.

Below there's an example of the data flow I have with my current setup (gyro data, but I can confirm it happens with the accel too):
```
$ rostopic hz /camera/gyro/sample -w 200
subscribed to [/camera/accel/sample]
average rate: 402.496
min: 0.000s max: 0.025s std dev: 0.00277s window: 200
average rate: 397.038
min: 0.000s max: 0.024s std dev: 0.00291s window: 200
average rate: 403.295
min: 0.000s max: 0.031s std dev: 0.00371s window: 200
no new messages
average rate: 79.371
min: 0.000s max: 2.119s std dev: 0.14974s window: 200
average rate: 402.004
min: 0.000s max: 0.023s std dev: 0.00257s window: 200
average rate: 397.415
min: 0.000s max: 0.021s std dev: 0.00235s window: 200
average rate: 402.149
no new messages
no new messages
no new messages

```

Also, I observed that whenever there's a loss of data (or exactly when there's `no new messages`), the stamp on the messages rolls-back in time. 
[image]

doronhi, 2020-11-16:
Tracked in DSO-16098

MartyG-RealSense, 2020-11-16:
@doronhi   Thanks very much for referencing the DSO number.

@zhizunhu  @ericpairet   Basically, a DSO means that Intel have a report filed about an issue, so it should get looked at by Intel at some point.  So it is probably best to wait for the outcome of the DSO, though I cannot offer a time estimate for that outcome.

ericpairet, 2020-11-17:
Thanks @doronhi for chipping in and @MartyG-RealSense for the details.

Is the DSO-16098 strictly related to the usage of the camera with embedded systems such as the Raspberry Pi? Asking as I cannot replicate the issue described above in either a Desktop or laptop machine; I cannot see any message loss in either setup (checked up to the first 100k messages). 

The inconsistency across platforms makes me think that this issue might result from bandwidth/processing limitations due to (a)~system without kernel patching (despite libUVC-backend) or (b)~system with limited computational power (the message loss is more significant as more sensors from the camera are enabled and the Pi gets busier).

Thinking that it might be some part of the installation that contributes to this issue, do we know of any image for the rasbperry pi, with librealsense and realsense-ros, that is not having this issue? I initially tried [Intel's official pre-built image for the Pi](https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras?-697455189.1592827118&_ga=2.21903602.813603083.1605425812-70032516.1601732816#section-2-3-preparing-the-sd-card) but I could not get the ROS wrapper working properly.

doronhi, 2020-11-17:
Thank you @ericpairet for the additional information and insight.
I'll make a point to include the Intel's Raspberry Pi image into the considerations.
I agree that frame drops are more common as resources drop low but why do time stamps go backwards require some inquiry.

MartyG-RealSense, 2020-11-23:
Hi @ericpairet   Do you require further assistance please?  Thanks!

ericpairet, 2020-11-24:
Hi @MartyG-RealSense, for what I understand, we cannot do much about frame drops and stamps going backwards. Just please update this issue if there is an outcome to the DSO-16098 or another solution to these problems is ever found.

MartyG-RealSense, 2020-11-24:
Hi @ericpairet   I will give this case an Enhancement tag to remind myself to keep this case open whilst DSO-16098 is active.

MartyG-RealSense, 2020-12-05:
Adding a note to keep this case open for a further time period.

etienn8, 2021-04-08:
Hi @MartyG-RealSense ! I encountered the same issue as @ericpairet. I'm also using a Raspberry Pi 4 and was wandering if you had any update about DSO-16098?

KlemenDEV, 2021-09-03:
Are there any updates on this?

MartyG-RealSense, 2021-09-03:
Hi @KlemenDEV   There are currently no updates available about this issue.

asimonov, 2022-01-04:
hello. I see the issue with same D455 on Intel laptop, but not on Jetson NX.
would be great to have an answer eventually why this happens and how to guarantee it is not happening.

MartyG-RealSense, 2026-10-31:
Case closed due to the internal investigation linked to this issue being closed in March 2024 because it could not be reproduced by the development team.

closed 2026-10-31 by MartyG-RealSense