realsenseai/realsense-ros #1473: Power consumption fluctuating and higher than expected while camera is disabled

PhilipAtLabrador, 2020-10-28:
We are using:
D435
Firmware 5.12.8.200
https://github.com/IntelRealSense/librealsense/releases/tag/v2.39.0
https://github.com/IntelRealSense/realsense-ros/commit/2ecaf739e1b4b6f97a287cb0193690dabdd68a51

We are using the enable/disable function to save power when we don't need data from the D435. One of the issues we are seeing is that the power consumption of the D435 fluctuates while it is disabled and is higher than we expect. We see that the camera oscillates between consuming ~135 ma and ~5 ma. It switches every few seconds and spends approximately 50% of it's time in each mode. Our goal is to get the power consumption as low as possible while still being able to resume normal operation quickly.

Other settings we are using:
6Hz
Visual preset 3(high accuracy)
emitter_on_off = true
inter_cam_sync_mode = 1

Our questions are:

1. Why is the camera switching modes?
2. Is the 135 ma mode necessary, or is it possible for it to stay in the 5 ma mode?

Thanks!

RealSenseSupport, 2020-10-29:
It seems that the similar topic was discussed in https://github.com/IntelRealSense/realsense-ros/issues/1195#issuecomment-633860772
Please take a look at several suggestions made by @doronhi. Please try unsubscribing the topic and releasing the nodes. 
Thank you.

PhilipAtLabrador, 2020-10-29:
Thanks for the response we are using the enable/disable function recently merged into the development branch with this pull request:
https://github.com/IntelRealSense/realsense-ros/pull/1313
My question is about this new functionality.

Unsubscribing, stopping the realsense node, and restarting it is not fast enough for our application. The new enable service is very nearly good enough, but we have this issue with it that we would like to improve.

doronhi, 2020-10-29:
Tracked by the team as DSO-16010.

RealSenseSupport, 2021-05-07:
Please check the latest ROS release for new start/stop sensor streaming. 
https://github.com/IntelRealSense/realsense-ros#available-services
We will close this case unless otherwise. Thank you.

closed 2021-05-07 by RealSenseSupport