realsenseai/realsense-ros #1476: Support for dynamic reconfigure of filters parameter

steviedale, 2020-10-29:
Is there a way to dynamically change the ```filters``` parameter? It looks like this is currently not supported. I was wondering if there is a known workaround for this? 

Essentially my team wants to keep the same realsense2_camera node running for several tasks. However, we need to reconfigure different parameters for each task, one of which is the ```filters``` parameter.

MartyG-RealSense, 2020-10-29:
Hi @steviedale   You could define the filters that you want to use in the roslaunch statement and then configure specific values for parameters supported for that filter after that.  This can be done at launch with rosparam definitions inside the launch file.  

Each parameter that can be set as a rosparam can also be set with dynamic reconfigure and vice versa.  The same parameter name is used for both.  Further details can be found in the link below.

https://github.com/IntelRealSense/realsense-ros/issues/1325

steviedale, 2020-10-29:
@MartyG-RealSense Thank you for your response, that is very helpful.

Follow up question: How would I disable a filter? For example, for one task I want to use the Decimate filter and in another task, I do not. I was hoping there would be an ```enable``` parameter, but all I see are the following parameters:
```
config: 
  bools: []
  ints: 
    - 
      name: "frames_queue_size"
      value: 16
    - 
      name: "filter_magnitude"
      value: 2
    - 
      name: "stream_filter"
      value: 1
    - 
      name: "stream_format_filter"
      value: 1
    - 
      name: "stream_index_filter"
      value: -1
  strs: []
  doubles: []
  groups: 
    - 
      name: "Default"
      state: True
      id: 0
      parent: 0
```

steviedale, 2020-10-29:
Perhaps I can set the ```frames_queue_size``` to zero to effectively disable it? I'm not sure what each of these parameters mean. I found [this page](https://dev.intelrealsense.com/docs/post-processing-filters) with some information about the filters, but it didn't mention any of these parameters.

MartyG-RealSense, 2020-10-29:
The new ROS2 wrapper for ROS2 Eloquent has enable and disable of filters and sensors during runtime planned for upcoming versions according to the wrapper release notes, if using ROS2 is an option for your project.

https://github.com/IntelRealSense/realsense-ros/releases/tag/3.1.1

If you need to use ROS1 then I once speculated that setting parameters to values that negate the effects of filters (e.g filter magnitude = 0) may be an equivalent to actually disabling them.

steviedale, 2020-10-29:
Thank you @MartyG-RealSense!

Another question (sorry!). How would I go about dynamically setting the `clip_distance` parameter? I don't see it in any of the set_parameter ROS services.

MartyG-RealSense, 2020-10-29:
I have seen it set as a launch **arg** sometimes:

`<launch>`
`<arg name="clip_distance" default="-2"/>`
`</launch>`

I have not seen it set with either a rosparam or dynamic reconfigure though, so could not confidently say if it can be done.

MartyG-RealSense, 2020-11-07:
Hi @steviedale   Do you still require assistance with this case, please?   Thanks!

steviedale, 2020-11-09:
@MartyG-RealSense we can close this issue. I decided to fork the repository and add in support for dynamic reconfigure of the `clip_distance` and `filters` parameters. Thank you for all the information and help!

MartyG-RealSense, 2020-11-09:
You are very welcome.   :) 

Great to hear about your success - thanks so much for the update!

closed 2020-11-09 by MartyG-RealSense