realsenseai/realsense-ros #1477: Realsense D435 with raspberrypi3, control_transfer error

AotukiSakuya, 2020-10-30:
| | |
|-------------|-------------|
|Platform|Raspberry Pi3|
|OS | Raspbian Stretch|
|Kernel Version | 4.19.66-v7+|
|Camera Model|D435|
|SDK Version|2.39.0|
|Firmware|5.12.8.200|
|ROS Version|Kinetic|

I have installed the librealsense2 packages and realsense-ros. Then, I try to run ```roslaunch realsense2_camera rs_camera.launch```. ROS Master is another PC(ubuntu 16.04). I checked these are connecting.
I got control_transfer error, and few seconds later topic has finished. I don't have any idea. 

i'm not good at English, so i'm sorry if it's s strange sentence.

Here is my output.

```
pi@costarica:~ $ roslaunch realsense2_camera rs_camera.launch
... logging to /home/pi/.ros/log/ce898e94-1a69-11eb-bb66-0492265ca40f/roslaunch-costarica-1985.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://10.40.1.80 :36893/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 6
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: False
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file:
 * /camera/realsense2_camera/clip_distance: -2.0
 * /camera/realsense2_camera/color_fps: 6
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 6
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type:
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: True
 * /camera/realsense2_camera/enable_infra2: True
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters:
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
[... 93 lines cut ...]
 30/10 15:57:00,438 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:00,489 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:00,539 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:00,590 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:00,641 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:00,691 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:00,742 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:01,489 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 30/10 15:57:01,539 WARNING [1774175264] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[camera/realsense2_camera-2] process has finished cleanly
log file: /home/pi/.ros/log/ce898e94-1a69-11eb-bb66-0492265ca40f/camera-realsense2_camera-2*.log
^C[camera/realsense2_camera_manager-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```
This is log file.
```
pi@costarica:~ $ cat .ros/log/latest/camera-realsense2_camera-2-stdout.log
[ INFO] [1604041015.427459471]: Loading nodelet /camera/realsense2_camera of type realsense2_camera/RealSenseNodeFactory to manager realsense2_camera_manager with the following remappings:
[ INFO] [1604041015.472003712]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1604041015.635584607]: waitForService: Service [/camera/realsense2_camera_manager/load_nodelet] is now available.
[ INFO] [1604041029.520003739]: Bond broken, exiting
```

MartyG-RealSense, 2020-10-30:
Hi @AotukiSakuya   Your English is good, don't worry.    :)

If the **control_transfer returned** is produced continuously then it can indicate a serious communication problem with the device.

Could you try adding the command below to the end of your roslaunch instruction please to see if it helps?

**initial_reset:=true**

This command performs a hardware reset of the camera at launch.

AotukiSakuya, 2020-10-31:
@MartyG-RealSense 
Thank you for replying!!
I add that command, but it gives no effect.

MartyG-RealSense, 2020-10-31:
The best librealsense match for ROS wrapper 2.2.17 is SDK 2.38.1.  If you are using librealsense 2.39.0, the best ROS wrapper version to match it with is the newest one at the time of writing this.   That is ROS wrapper **2.2.18**.

https://github.com/IntelRealSense/realsense-ros/releases/tag/2.2.18

AotukiSakuya, 2020-10-31:
I updated, but nothing changed.
Is there any possibility my platform causes problems?

MartyG-RealSense, 2020-10-31:
Raspberry Pi 3 only has 1 GB of memory, so performance may slow once that real memory is used up and the Pi starts using the virtual memory of the swapfile instead.

Also, Pi 3 has USB 2 only (no USB 3 like the Pi 4 model has) and so the capabilities of the camera will be more limited due to the lower bandwidth of a USB2 connection.

https://www.intelrealsense.com/usb2-support-for-intel-realsense-technology/

You could try using 15 FPS instead of 6 FPS, as there is a greater chance of errors occurring at 6 FPS.

AotukiSakuya, 2020-11-03:
Using 15 FPS couldn't solve my problems, but i found good news.

When I tested `librealsense/wrappers/python/examples/align-depth2color.py`, no images were drawing even if Realsense-viewer displayed images.  This means Realsense get right data. Now I think opencv or other libraries cause these error.

MartyG-RealSense, 2020-11-03:
Hi @AotukiSakuya  I'm pleased to hear that you made progress in diagnosing your problem.   Please do keep me updated about your progress and feel free to ask any new questions that come to mind.   Good luck!

closed 2020-11-17 by MartyG-RealSense