realsenseai/realsense-ros #1480: D435i

sayyidabeegam, 2020-11-02:
**This is showing when I launch 
roslaunch realsense2_camera rs_camera.launch**

[ INFO] [1604320706.611571036]: Device with physical ID 2-1.4-4 was found.
[ INFO] [1604320706.611584299]: Device with name Intel RealSense D435I was found.
[ INFO] [1604320706.612108270]: Device with port number 2-1.4 was found.
[ INFO] [1604320706.613937119]: getParameters...
[ INFO] [1604320706.668293424]: setupDevice...
[ INFO] [1604320706.668321254]: JSON file is not provided
[ INFO] [1604320706.668334749]: ROS Node Namespace: camera
[ INFO] [1604320706.668350293]: Device Name: Intel RealSense D435I
[ INFO] [1604320706.668364177]: Device Serial No: 939622071716
[ INFO] [1604320706.668377354]: Device physical port: 2-1.4-4
[ INFO] [1604320706.668389360]: Device FW version: 05.12.07.100
[ INFO] [1604320706.668401953]: Device Product ID: 0x0B3A
[ INFO] [1604320706.668415539]: Enable PointCloud: Off
[ INFO] [1604320706.668426481]: Align Depth: Off
[ INFO] [1604320706.668436209]: Sync Mode: Off
[ INFO] [1604320706.668453828]: Device Sensors: 
[ INFO] [1604320706.683095835]: Stereo Module was found.
[ INFO] [1604320706.686958874]: RGB Camera was found.
[ INFO] [1604320706.687069098]: Motion Module was found.
[ INFO] [1604320706.687091069]: num_filters: 0
[ INFO] [1604320706.687104203]: Setting Dynamic reconfig parameters.
[ INFO] [1604320707.148347117]: Done Setting Dynamic reconfig parameters.
[ INFO] [1604320707.148671195]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1604320707.149067517]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1604320707.149090050]: setupPublishers...
[ INFO] [1604320707.151022750]: Expected frequency for depth = 30.00000
[ INFO] [1604320707.176197618]: Expected frequency for color = 30.00000
[ INFO] [1604320707.199192191]: setupStreams...
[ INFO] [1604320707.212007627]: insert Depth to Stereo Module
[ INFO] [1604320707.212053599]: insert Color to RGB Camera
[ INFO] [1604320707.253098934]: SELECTED BASE:Depth, 0
[ INFO] [1604320707.263290341]: RealSense Node Is Up!
 02/11 18:08:27,381 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/11 18:08:27,432 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/11 18:08:27,583 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/11 18:08:27,635 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 02/11 18:08:27,686 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/11 18:08:27,736 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/11 18:08:27,787 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/11 18:08:27,939 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 02/11 18:08:27,989 WARNING [139938305644288] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11

MartyG-RealSense, 2020-11-02:
Hi @sayyidabeegam   If you only receive a small number of **control_transfer returned** warnings at launch, or at a rate of one or two per minute, then it is safe to ignore them.  If this particular warning is being generated continuously though then it can indicate that there is a serious communication problem with the device.

As your log does not indicate that the launch process exited, could you check whether the topics are correctly published in RViz despite the warnings please?

sayyidabeegam, 2020-11-02:
Topics are publishing correctly.
Thank you

closed 2020-11-02 by sayyidabeegam

MartyG-RealSense, 2020-11-02:
You are very welcome   :)