realsenseai/realsense-ros #1482: Mapping an RGB image point to a Depth image point

rafaelRinconFo, 2020-11-02:
Greetings guys.

I'm currently developing an application and I need to click a point on the RGB image, and after that, obtaining its depth value. The problem with the pyrealsense2 library is the fact that you map the entire depth image into the RGB image and that would be inefficient for me.

I only need to use coordinates of the RGB pixel and looking its depth value on the depth image, the entire image would be redundant. 

It is not possible to obtain directly this mapping, both images have the same resolution but the Stereo cameras have a wider FOV when compared to the RGB camera (I'm currently using the D435i camera).

Thank you guys in advance for your answer.

MartyG-RealSense, 2020-11-02:
Hi @rafaelRinconFo   I hope that the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/7624

rafaelRinconFo, 2020-11-02:
Thank you, luckily, when I was searching for documentation, I found out that there is a topic `/camera/aligned_depth_to_color/image_raw ` That already maps the depth and RGB image in the Wrapper I'm using. Thank you so much anyway for the information you provided @MartyG-RealSense

closed 2020-11-02 by rafaelRinconFo

MartyG-RealSense, 2020-11-02:
Thanks very much for the update @rafaelRinconFo  - I'm pleased to hear that you found a solution that meets your needs.