realsenseai/realsense-ros #1483: control_transfer returned error, index: 768, error: No data available, number: 61

sdkaplan, 2020-11-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Ubuntu 18.04 LTS  | 
| Kernel Version (Linux Only)    | 4.9.140-tegra   | 
| Platform                                 | Nvidia Jetson Xavier  |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C++                          | 

when I run `roslaunch rs_camera.launch` I get the following errors:
` 02/11 18:42:56,885 WARNING [546962383232] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61`

When I open the realsense-viewer I get the following error:
`UNKNOWN in rs2_create_device(info_list:0x55a7a85640, index:0):
failed to set power state`

When I run `rs-enumerate-devices` I get the following error:
` 02/11 18:46:45,465 ERROR [547522572064] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
Could not create device - failed to set power state . Check SDK logs for details`

Finally when I run `dmesg --follow` I get  the following results:
`[27624.311310] input: Intel(R) RealSense(TM) Depth Ca as /devices/3610000.xhci/usb2/2-3/2-3:1.0/input/input27
[27624.311976] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[27624.311986] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[27624.314470] uvcvideo: Unable to create debugfs 2-6 directory.
[27624.314685] uvcvideo 2-3:1.3: Entity type for entity Processing 7 was not initialized!
[27624.314866] uvcvideo 2-3:1.3: Entity type for entity Extension 8 was not initialized!
[27624.315003] uvcvideo 2-3:1.3: Entity type for entity Camera 6 was not initialized!
[27624.376319] hid-sensor-hub 0003:8086:0B3A.0016: No report with id 0xffffffff found`

Any idea what my error could be?

MartyG-RealSense, 2020-11-03:
Hi @sdkaplan   Historically, **Failed to set power state** is most likely to occur on a Jetson Xavier when there is more than one camera attached.   So may I begin please by asking whether there is one camera attached or more than one?  Thanks very much!

sdkaplan, 2020-11-03:
Hello, thanks for your response. I only have one device attached. I tried changing the realsense-ros version to 2.2.18 and including an initial reset

MartyG-RealSense, 2020-11-03:
Wrapper version 2.2.17 would be a better match for SDK 2.38.1, as the 2.2.18 wrapper is aimed at SDK 2.39.0.  Your problems with the camera do not seem to be limited to ROS, however.

In the RealSense Viewer, if you launch it without the camera plugged in, does the **Failed to set power state** message appear when you insert the camera into the USB port and you have no camera details displayed in the options panel?

sdkaplan, 2020-11-04:
When I launch the real sense viewer without the camera plugged in I get the following error: 
Timestamp: 1604511275588.660645
Severity: Warn
Description: Multiple realsense udev-rules were found! :
1:/etc/udev/rules.d/99-realsense-libusb.rules
2: /lib/udev/rules.d/60-librealsense2-udev-rules.rules
Make sure to remove redundancies!

Today when I plug the camera in I have an error messagemthat I cannot read, but the camera shows plugged in:
[image]

sdkaplan, 2020-11-04:
[image]

I noticed a bunch of error codes in the terminal so I repeated the same steps again and now the camera is not showing with he failed to set powerstate error. 

Do you think thisd is an error with the version of librealsense on my jetson? If so can you help me figure out which version of librealsense and realsense-ros would be more stable on my machine?

MartyG-RealSense, 2020-11-04:
Multiple udev rules (which handle devices) can occur if librealsense is installed both with packages (the 'distribution' method) and by building librealsense from source code.  It is safe to remove the rules installed in the **/etc/udev** folder.  The rules in the **/lib/udev** folder should not be deleted though, as doing so can break package integrity.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

sdkaplan, 2020-11-04:
I have deleted the rules in /etc/udev and when I restarted realsense-viewer I don't get errors in the viewer but I get the following errors in the terminal:

 04/11 11:57:40,158 WARNING [547925053888] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 04/11 11:57:40,211 WARNING [547925053888] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No such device, number: 13
 04/11 11:57:40,212 WARNING [547925053888] (messenger-libusb.cpp:92) usb_request_queue returned error, endpoint: 87 error: No such device, number: 13
 04/11 11:57:40,212 ERROR [547675435456] (error-handling.cpp:68) Error during polling error handler: usb device disconnected
 04/11 11:57:41,212 ERROR [547925053888] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_NO_DEVICE
 04/11 11:57:41,213 ERROR [547675435456] (error-handling.cpp:68) Error during polling error handler: failed to set power state
 04/11 11:57:43,352 WARNING [547599901120] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 04/11 11:57:43,729 WARNING [547599901120] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 04/11 11:57:43,792 WARNING [547599901120] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 04/11 11:57:43,845 WARNING [547599901120] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 04/11 11:57:47,227 ERROR [547675435456] (error-handling.cpp:68) Error during polling error handler: get_xu(id=7) failed! Last Error: Success

MartyG-RealSense, 2020-11-04:
Do you still get the errors if you include **initial_reset:=true** in your roslaunch instruction please?

sdkaplan, 2020-11-04:
[image]

I get the errors at the bottom of the screenshot and the errors warnings at the top of the screenshot in yellow

MartyG-RealSense, 2020-11-04:
Is RViz publishing the topics despite the messages?

sdkaplan, 2020-11-04:
Yes it is, do you think that the problem is fixed then?

MartyG-RealSense, 2020-11-04:
If the topics are publishing correctly in RViz and performance does not seem to be degraded then it is likely safe to ignore the **control_transfer returned** errors.  If they only appear at launch, or appear one or two times a minute after launch, then they should not be a problem.

MartyG-RealSense, 2020-11-19:
Case closed due to no further comments received.

closed 2020-11-19 by MartyG-RealSense

ranjitkathiriya, 2020-12-17:
Hi Marty,

I am facing the same issue and I have tried these steps but I am not able to troubleshoot where I am wrong. Can you please help me as well? Below I am stating my error:

roslaunch realsense2_camera rs_rgbd.launch filters:=colorizer
... logging to /home/kuka-ai/.ros/log/7eab505c-404c-11eb-8825-4fa03bac19fa/roslaunch-kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU-17402.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:46357/

SUMMARY
========

PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
 * /camera/realsense2_camera/clip_distance: -1.0
 * /camera/realsense2_camera/color_fps: 30
 * /camera/realsense2_camera/color_frame_id: camera_color_frame
 * /camera/realsense2_camera/color_height: 480
 * /camera/realsense2_camera/color_optical_frame_id: camera_color_opti...
 * /camera/realsense2_camera/color_width: 640
 * /camera/realsense2_camera/depth_fps: 30
 * /camera/realsense2_camera/depth_frame_id: camera_depth_frame
 * /camera/realsense2_camera/depth_height: 480
 * /camera/realsense2_camera/depth_optical_frame_id: camera_depth_opti...
 * /camera/realsense2_camera/depth_width: 640
 * /camera/realsense2_camera/device_type: 
 * /camera/realsense2_camera/enable_accel: False
 * /camera/realsense2_camera/enable_color: True
 * /camera/realsense2_camera/enable_depth: True
 * /camera/realsense2_camera/enable_fisheye1: False
 * /camera/realsense2_camera/enable_fisheye2: False
 * /camera/realsense2_camera/enable_fisheye: False
 * /camera/realsense2_camera/enable_gyro: False
 * /camera/realsense2_camera/enable_infra1: False
 * /camera/realsense2_camera/enable_infra2: False
 * /camera/realsense2_camera/enable_infra: False
 * /camera/realsense2_camera/enable_pointcloud: False
 * /camera/realsense2_camera/enable_pose: False
 * /camera/realsense2_camera/enable_sync: True
 * /camera/realsense2_camera/filters: colorizer
 * /camera/realsense2_camera/fisheye1_frame_id: camera_fisheye1_f...
 * /camera/realsense2_camera/fisheye1_optical_frame_id: camera_fisheye1_o...
 * /camera/realsense2_camera/fisheye2_frame_id: camera_fisheye2_f...
 * /camera/realsense2_camera/fisheye2_optical_frame_id: camera_fisheye2_o...
 * /camera/realsense2_camera/fisheye_fps: 30
 * /camera/realsense2_camera/fisheye_frame_id: camera_fisheye_frame
 * /camera/realsense2_camera/fisheye_height: 480
 * /camera/realsense2_camera/fisheye_optical_frame_id: camera_fisheye_op...
 * /camera/realsense2_camera/fisheye_width: 640
 * /camera/realsense2_camera/gyro_fps: 400
 * /camera/realsense2_camera/gyro_frame_id: camera_gyro_frame
 * /camera/realsense2_camera/gyro_optical_frame_id: camera_gyro_optic...
 * /camera/realsense2_camera/imu_optical_frame_id: camera_imu_optica...
 * /camera/realsense2_camera/infra1_frame_id: camera_infra1_frame
 * /camera/realsense2_camera/infra1_optical_frame_id: camera_infra1_opt...
[... 73 lines cut ...]
[ INFO] [1608198286.715677651]: depth stream is enabled - width: 640, height: 480, fps: 30, Format: Z16
[ INFO] [1608198286.716179957]: color stream is enabled - width: 640, height: 480, fps: 30, Format: RGB8
[ INFO] [1608198286.717695282]: setupPublishers...
[ INFO] [1608198286.720227272]: Expected frequency for depth = 30.00000
[ INFO] [1608198286.739280145]: Expected frequency for color = 30.00000
[ INFO] [1608198286.748967745]: Expected frequency for aligned_depth_to_color = 30.00000
[ INFO] [1608198286.763182554]: setupStreams...
[ INFO] [1608198286.775722738]: insert Depth to Stereo Module
[ INFO] [1608198286.775827866]: insert Color to RGB Camera
[ INFO] [1608198286.854717784]: SELECTED BASE:Depth, 0
[ INFO] [1608198286.872840084]: RealSense Node Is Up!
 17/12 09:44:47,003 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 17/12 09:44:47,054 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 09:44:47,110 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 09:44:47,285 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 09:44:47,336 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[... cut ...]
 17/12 09:44:47,942 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 09:45:29,727 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 300, error: No data available, number: 3d
 17/12 09:45:29,778 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
 17/12 09:53:20,198 WARNING [140544013498112] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11