realsenseai/realsense-ros #1484: intel D415 CompressedDepth extremely delayed/no message

Misticfury, 2020-11-03:
im doing remote mapping on a raspbarry pi 4 with the Realsense d415 and am using the rgbd_sync to compress  the video and depth but the compressedDepth is coming in extremely delayed and/or no message when ever i do a rostopic hz. because of that my rtabmap quickly loses itself if i move the camera.

from my realsense i also keep getting this error msg:
03/11 03:10:06,726 ERROR [281471872913792] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 132

this is what im launching to do the compression

  ```
<arg name="rate"         default="5"/>
  <arg name="decimation"   default="3"/>
  <arg name="depth_scale"  default="1"/>
  <arg name="approx_sync"  default="false" /> <!-- true for freenect driver -->
  <arg name="rgbd_sync"    default="true"/>
  <arg name="node_start_delay" default="15.0" />

  <!-- Use same nodelet used by Freenect/OpenNI -->
  <group ns="camera">
    <node if="$(arg rgbd_sync)" pkg="nodelet" type="nodelet" name="rgbd_sync" args="load rtabmap_ros/rgbd_sync camera_nodelet_manager" output="screen"
     launch-prefix="bash -c 'sleep $(arg node_start_delay); $0 $@' ">

      <param name="compressed_rate"  type="double" value="$(arg rate)"/>
      <param name="decimation"       type="double" value="$(arg decimation)"/>
      <param name="depth_scale"      type="double" value="$(arg depth_scale)"/>
      <param name="approx_sync"      type="bool"   value="$(arg approx_sync)"/>

      <remap from="camera_nodelet_manager/load_nodelet" to="camera_nodelet_manager/load_nodelet"/>
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>

      <remap from="rgbd_image"      to="rgbd_image"/>
    </node>
  </group>
```

and these are the argument for my rtabmap.launch
   
```
  <!-- Choose between depth and stereo, set both to false to do only scan -->      
  <arg name="stereo"                    default="false"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
 
  <!-- Choose visualization -->
  <arg name="rtabmapviz"              default="false" /> 
  <arg name="rviz"                    default="false" />
  
  <!-- Localization-only mode -->
  <arg name="localization"            default="false"/>
  
  <!-- sim time for convenience, if playing a rosbag -->
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>
  
  <!-- Corresponding config files -->
  <arg name="cfg"                     default="" /> <!-- To change RTAB-Map's parameters, set the path of config file (*.ini) generated by the standalone app -->
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />
  
  <arg name="frame_id"                default="camera_link"/>     <!-- Fixed frame id, you may set "base_link" or "base_footprint" if they are published -->
  <arg name="odom_frame_id"           default=""/>                <!-- If set, TF is used to get odometry instead of the topic -->
  <arg name="map_frame_id"            default="map"/>
  <arg name="ground_truth_frame_id"   default=""/>     <!-- e.g., "world" -->
  <arg name="ground_truth_base_frame_id" default=""/>  <!-- e.g., "tracker", a fake frame matching the frame "frame_id" (but on different TF tree) -->
  <arg name="namespace"               default="rtabmap"/>
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="queue_size"              default="10"/>
  <arg name="wait_for_transform"      default="0.2"/>
  <arg name="args"                    default="--delete_db_on_start"/>              <!-- delete_db_on_start, udebug -->
  <arg name="rtabmap_args"            default="$(arg args)"/>   <!-- deprecated, use "args" argument -->
[... 64 lines cut ...]
  <arg name="scan_cloud_assembling_noise_radius"   default="0.0"/>    <!-- 0=disabled -->
  <arg name="scan_cloud_assembling_noise_min_neighbors"   default="5"/>
  
  <arg name="subscribe_user_data"      default="false"/>             <!-- user data synchronized subscription -->
  <arg name="user_data_topic"          default="/user_data"/>
  <arg name="user_data_async_topic"    default="/user_data_async" /> <!-- user data async subscription (rate should be lower than map update rate) -->
  
  <arg name="gps_topic"                default="/gps/fix" />         <!-- gps async subscription -->
  
  <arg name="tag_topic"                default="/tag_detections" />  <!-- apriltags async subscription -->
  <arg name="tag_linear_variance"      default="0.0001" />
  <arg name="tag_angular_variance"     default="9999" />             <!-- >=9999 means ignore rotation in optimization, when rotation estimation of the tag is not reliable -->
  
  <!-- These arguments should not be modified directly, see referred topics without "_relay" suffix above -->
  <arg if="$(arg compressed)"     name="rgb_topic_relay"           default="$(arg rgb_topic)_relay"/>
  <arg unless="$(arg compressed)" name="rgb_topic_relay"           default="$(arg rgb_topic)"/>
  <arg if="$(arg compressed)"     name="depth_topic_relay"         default="$(arg depth_topic)_relay"/>
  <arg unless="$(arg compressed)" name="depth_topic_relay"         default="$(arg depth_topic)"/>
  <arg if="$(arg compressed)"     name="left_image_topic_relay"    default="$(arg left_image_topic)_relay"/>
  <arg unless="$(arg compressed)" name="left_image_topic_relay"    default="$(arg left_image_topic)"/>
  <arg if="$(arg compressed)"     name="right_image_topic_relay"   default="$(arg right_image_topic)_relay"/>
  <arg unless="$(arg compressed)" name="right_image_topic_relay"   default="$(arg right_image_topic)"/>
  <arg if="$(arg rgbd_sync)"      name="rgbd_topic_relay"          default="$(arg rgbd_topic)"/>
  <arg unless="$(arg rgbd_sync)"  name="rgbd_topic_relay"          default="$(arg rgbd_topic)_relay"/>
```

MartyG-RealSense, 2020-11-03:
Hi @Misticfury   As the value of **compressed_rate** is based on your defined value of **arg rate** (which you have set to '5' by default), do you get better results if you set it to a number between 1 and 3?

Misticfury, 2020-11-03:
so i found a solution to it but i just realize that the rgbd_sync isn't doing anything i don't know if im using it correctly. correct me if im wrong so would setting the compresses_rate to 5 limit the fps to 5?

this is what i added in the launch file and now my compressedDepth is a stable 11fps
  ```
  <param name="/camera/color/image_raw/compressed/format" type="string" value="jpeg" />
  <param name="/camera/color/image_raw/compressed/jpeg_level"  type="int" value="1" />
  <param name="/camera/aligned_depth_to_color/image_raw/compressedDepth/format" type="string" value="png" />
  <param name="/camera/aligned_depth_to_color/image_raw/compressedDepth/png_level"  type="int" value="0" />
```

MartyG-RealSense, 2020-11-03:
RTABMAP is not one of my specialist knowledge subjects.  Apparently though, compressed_rate limits the output rate of the compressed topic, according to the link below.

https://github.com/introlab/rtabmap_ros/issues/458#issuecomment-684977247

MartyG-RealSense, 2020-11-11:
Hi @Misticfury   Do you still require assistance with this case pleae?  Thanks!

closed 2020-11-18 by Misticfury